moggieuk/Happy-Hare #319: Random communication loss with MCU 'mmu'

ksummers92, 2024-06-19:
I am using a SKR Pico as the mmu MCU for a Trad Rack. Recently, I have been encountering seemingly random communication losses with the mmu MCU during print regardless of whether filament changes are involved or not. I have using extruder sync with a belay and it works great when it isn't failing to print due to communication loss. I have attached the mmu log and the link to the klippy log is here as the file size is too large for Github: https://drive.proton.me/urls/SC7W50ZGF8#9wKTdAcW9zVP
[mmu 4.log](https://github.com/user-attachments/files/15894142/mmu.4.log)

The only correlation that I can see in the logs are the it appears to occur in conjunction with a _MMU_ENCODER_RUNOUT command. I am not sure they are related or why that would cause this issue.

moggieuk, 2024-06-26:
The communication timeout is related to the ability of the rpi to keep up or a hardware cabling issue.  Couple of things to check for the rpi load are:
1) Turn `mmu_log_file_level: 2` to reduce the volume.
2) Read https://github.com/moggieuk/Happy-Hare/wiki/Troubleshooting-and-Common-Issues#-timer-too-close

I know you don't have a TTC error but the last part of this description could be the cause of a rpi slow down.

moggieuk, 2024-07-10:
Closing this Issue based on inactivity.

closed 2024-07-10 by moggieuk