moggieuk/Happy-Hare #324: Buffer Array

alidaf, 2024-06-24:
In your instructions you state: "I run mine with a standalone Raspberry Pi attached to my buffer array and can control multiple MMU's with it." What is a buffer array? I'm trying to determine whether I need to replace the display on my Voron 2.4 with a new touch display but this implies that I don't need to.

moggieuk, 2024-06-26:
Buffer array is simply a line of filament buffers (like ERCT).  Mine are in a cabinet and I attached my rpi with KlipperScreen to that because it sits above the printer.

KlipperScreen can be attached to rpi on the printer OR can be on a separate rpi.  I prefer separate so I can control all my printers from one panel..

alidaf, 2024-06-26:
Makes sense. Thank you.

closed 2024-06-26 by alidaf