moggieuk/Happy-Hare #325: Just pull out?

Durahl, 2024-06-26:
Over the years I've tried all the Tip Forming BS and quite honestly the best Tips I've gotten since ever ( minus the harmless Whisks ) have been the ones where I had to pull the Filaments out by Hand in a single Move at a "Humane" 150-200mm/s.

Which Settings would have to be edited to disable all of the Initial Pull, Parking, Dipping, etc... Stuff and just have a single Pull Move until it clears the Extruder for the Gear Motor to take over? 🤔

Ecaze81, 2024-07-18:
I'm a fan of this type of solution, if I simply go back I have a valid tip, but with the macro it makes me stuck on the dragon or dragonfly.
Did you write your own macro?
can I make a simple unload macro?
`[gcode_macro UNLOAD_FILAMENT]
gcode:
    {% set speed = params.SPEED|default(300) %}
    {% set max_velocity = printer.configfile.settings['extruder'].max_extrude_only_velocity %}
    SAVE_GCODE_STATE NAME=unload_state
    G91
    G92 E0
    G1 E25 F{speed} # purge
    G1 E-50 F600 # fast-unload
    RESTORE_GCODE_STATE NAME=unload_state`

Ecaze81, 2024-07-18:
I don't understand what I'm doing wrong but he doesn't want it
```
MMU issue detected whilst out of a print
Reason: Unload sequence failed: Filament tip forming macro 'MY_FORM_TIP' not found.
Occured when changing tool: T0 > T1
```
[image]

closed 2024-11-28 by github-actions[bot]