moggieuk/Happy-Hare #327: Option to automatically (re-)home selector in case of error loading filament at gate

Belstrekkie, 2024-06-30:
It rarely happens to me, but if I get a filament loading error (not enough movement detected at the gate) it is solved in 100% of the cases by homing the selector and reload. Instead of homing the selector before very tool change like @rwijnhov suggested earlier, wouldn't it be possible to add the option to automatically home the selector in case of this type of loading error and try loading again after homing, before pausing with an MMU issue?

moggieuk, 2024-07-08:
It would be possible but I'm not sure always what you want to do.  A loading error could have caused the gate to block so homing may not be possible. Firstly, why exactly do you get the error of "not enough movement" -- that suggests that the filament is not being picked up and likely because the servo is not gripping the filament correctly or that you gate is not aligned.  I'm guessing the latter because that's the only thing re-homing would help.

Can you explain more what is physically going on...?

Belstrekkie, 2024-07-08:
It is indeed an alignment issue, but so minimal that isn't clearly visible (perhaps the usage of a filament cutter makes the system a little bit more sensitive for small mis alignments because of the blunt tip after cutting). It happens every now and than, but if it happens, it does so in the middle of the night of course leading to a massive disappointment in the morning ;). I never had any other causes for this error, so in my case an option to automatically re-home and retry loading when this error happens would be great. I'm also not sure which other causes could lead to this error. If the filament isn't picked up at all, you get an empty gate error. If the servo isn't grapping the filament firmly, this most likely will already happen during the initial check if filament is present at the start of a print and much less likely to occur suddenly during a print? 
If the selector is blocked and unable to re-home, than the 'selector didn't find home (check for blockage?)' would help to solve the issue. Besides from re-homing, the retry of loading already exists and this would be just another retry, so no real change there I think.

moggieuk, 2024-07-15:
Ok, so you are using EREC?  Hmm, is it possible that powerful servo movement nudges the selector? 

One suggestion is to increase or remove the `hold_current` for selector motor.  I typically don't recommend this because of the extra heat it generates in that stepper but in this case maybe you want to increase a little.

moggieuk, 2024-07-31:
I really don't think re-homing is the answer.  The is absolutely no reason the selector should move during a print. Imagine what would happen if you X or Y axis "moved"?   I would check two things:
1. the selector drive belt on ERCF isn't tool loose.
2. the selector slides smoothly.

BTW You would get better help of the ERCF forums because I do think this may be a build tuning issue.

closed 2024-11-27 by github-actions[bot]