moggieuk/Happy-Hare #332: Spoolman integration enhancements

CooperGerman, 2024-07-02:
Hi, 
firstly very big thanks for this outstanding work! I just switched my ercf v1.1b to HH and must say i am very happy with the proposed features as for now.

I don't know if this is the place to discuss this but i have several thoughts about the spoolman integration and would like to discuss some (and maybe contribute). See my [fork of moonraker](https://github.com/CooperGerman/moonraker_spoolman_enhancement)

In my previous setup i had added pre print checks to the ercf software based on the spoolman.py compoment enhancement.
- **initialize and keep track of which slot (gate) has wich spool_id**
- activate, clear, get ... slots 
- **check filament usage before printing (added metadata fields to be parsed in metadata.py)**
- generate swap table according to found filament and requested print
- enable printing with only one filament even though i have 9 gates
- remap tools automatically to correct gates with swap_table
- **add persistence (with variables.cfg** 

The bold features are the one i managed to find in HH and i was wondering if it was planned to add the features i'm currently missing. I'm willing to try and rework my previous fork of moonraker in order to accomodate HH.

For future discussion, i found out about this [PR](https://github.com/SoftFever/OrcaSlicer/pull/4771) in OrcaSlicer that might help for a very nice user experience. I learned that it is possible to use the "extra" field in order to store custom information, i was hoping to use it to store the gate mapping for the current machine there. For now i was using the `location` field with something like `<hostname>:<slot_nb>`.

Thanks again for all this great (and very well documented) work!

moggieuk, 2024-07-08:
Sorry for the delay -- been traveling.

In reading your list and trying to infer the exact need, it sounds like almost all of this is available now, sans a little bit of custom macro to setup the TTG (tool-to-gate map) automatically prior to print.

Have a read of: 
https://github.com/moggieuk/Happy-Hare/wiki/Tool-and-Gate-Maps

I think the only thing needed would be to tack on the spoolman id to the "Slicer tool map".

Then the macro to setup TTG would be easy.

Let me know

CooperGerman, 2024-07-08:
Hi, thanks for the reply, I'm not in a hurry :wink: 

I've been carefully reading all your doc and code, and started adding some features into my own fork of HH.

I primarily wanted to keep the previous functionality i had working on my custom ercf software environment:
- **storing the gate mapping remotely (in spoolman)** 

I understand a lot can be achieved through macros but i was willing to offer my help in order to develop a more _builtin_ version.

I added a `enable_remote_gate_map` to the `mmu_parameters.cfg` file which enables setting the mmu_gate_map to be persistent in the spoolman db through the usage of a `machine_name` field (set to printer_hostname for now) and a `mmu_gate_map` field used to store to the gate number the spool is mapped to for machine `machine_name`.

[My PR](https://github.com/moggieuk/Happy-Hare/pull/335#issue-2395845620) for now offers the above mentioned but is still in draft.

I really would like to add the following:
- recognize that a filament for the print is not on correct gate and issue the correct ttg remap command
- handle prints with less filaments set as HH expects, leverage the previous feature to auto remap
- add convenience checks in the `MMU_START_CHECKS` on filament usage and naming 

Thank you for taking the time to discuss this with me, have a nice day !

CooperGerman, 2024-07-09:
#335 has been updated with more detailed explanation of the implemented features

fluetke, 2024-10-08:
thank you very much for your awesome work, may I just add one little idea to the spoolman enhancements.

Currently the slicer toolmap is generated from the gcode and colors of each tool are included depending on the slicers configuration, these colors are then displayed in the "extruder" section in mainsail. 
Would it be possible to instead use the info stored in spoolman, to display the tool colors in the extruder section? 

It seems to me that the most up to date information on tool-colors and materials would come from the MMU_GATE_MAP (when spool-ids are assigned to all the gates), instead of the theoretical values stored in the slicers gcode (especially given the current issues with extruder_colour, vs. filament_colour).

Maybe I'm missing something, but I couldn't find info on how to achieve that.

CooperGerman, 2024-10-09:
Hi, let me make sure i understand your point. I understand you would like for colors to be fetched from the spoolman db when being displayed in mainsail instead of the colors beeing retrieved from the slicer (metadata from the gcode file) ?

To me this looks like it is beeing supported as for now. You can indeed enable the spoolman feature in push/pull or readonly mode and it would then use attributes for the spools fetched from spoolman.

```
#
#                    | Activate/  | Fetch filament attributes | Filament gate    | Filament gate     |
#   spoolman_support | Deactivate | attributes from spoolman  | assignment shown | assignment pulled |
#                    | spool?     | based on spool_id?        | in spoolman db?  | from spoolman db? |
#   -----------------+------------+---------------------------+------------------+-------------------+
#        off         |     no     |           no              |        no        |        no         |
#        readonly    |     yes    |           yes             |        no        |        no         |
#        push        |     yes    |           yes             |        yes       |        no         |
#        pull        |     yes    |           yes             |        yes       |        yes        |
```
Also see : [this section of the doc](https://github.com/moggieuk/Happy-Hare/wiki/Spoolman-Support#---gate-map-and-spool-id) for info about GATE_MAP
and [this one](https://github.com/moggieuk/Happy-Hare/wiki/Mainsail-Fluidd-Integration#---extruderfilament-color) for infor about extruder/filament color being displayed in mainsail

As i understand your point you would want to have the displayed colors in mainsail to match the ones set in the spoolman db ? 

_Side note: if slicer and db are not set up with matching names, colors etc... then the automap feature will be trickier to set up._

fluetke, 2024-10-09:
>  I understand you would like for colors to be fetched from the spoolman db when being displayed in mainsail instead of the colors beeing retrieved from the slicer (metadata from the gcode file) ?

Yes, currently I map gates to filaments by doing a "MMU_GATE_MAP GATE=1 SPOOLID=123" in the klipper console, whenever I change the filament in the MMU. 

Before using the Spoolman integration I used "MMU_GATE_MAP GATE=1 MATERIAL=PLA COLOR=112233", that apparently updated the gate map successfully with materials and colors, but the colors of the gate map did not show up on the tools in Mainsails extruder section.

Now, when I assign the spoolids to gates using the console command, the colors still didn't show up in the extruder section, hence my assumption that this needed to be implemented. 

If I understand your comment and the sections linked correctly, I might have just missed some steps and pages of the documentation and this is already working, I was just too blind to spot it before. 

Thank you so much for the clarification and the pointers in the right direction.

SamueldaCostaAraujoNunes, 2025-01-20:
Hi guys! I've been thinking a bit about the integration between Klipper and Spoolman. I know that today, when starting a print, the slicer colors are displayed in the tools/extruders menu. This guy here:

[image]

However, I miss the gate color manager, which we have in KlipperScreen with happyhare's mod.

closed 2026-08-21 by CooperGerman