moggieuk/Happy-Hare #333: Reverse gear stepper based on gate selection - 3DChameleon Integration

glennkaufman, 2024-07-04:
I am looking for a way to reverse filament direction based on active tool selected. T0 and T1 use the default loading and unloading procedure. However, when switched to T2 and T3, the filament runs on the opposite side of the gear stepper and thus reverses all procedures. I would need to change the DIRECTION_LOAD = 1 to DIRECTION_LOAD = -1 and DIRECTION_UNLOAD = -1 to DIRECTION_UNLOAD = 1.

moggieuk, 2024-07-08:
There was another user to modified HH to do this but I'm forgetting their name.  It requires getting into the HH python logic `mmu_toolhead.py`.    You might be able to track them down through the Voron Discord channels..

moggieuk, 2024-07-08:
I added to my TODO list because supporting 3DChameleon would be interesting, but are there other changes that would be necessary?

glennkaufman, 2024-07-29:
Sorry for the delay. The reverse spooling is the big one. The selector would use sensor less homing and each gate is basically a quarter turn of the selector stepper. Default is at gate 0. Tip shaping is needed with no tool head cutter. I ended up just setting the gates manually after setup.

moggieuk, 2026-10-21:
Closing this as 3D Chameleon is now a supported MMU type in Happy Hare v4.

In menuconfig choose "3D Chameleon" as the MMU type. It uses the `RotarySelector`, which reverses the gear direction per gate via `selector_gate_directions` in `mmu_parameters.cfg` (default `1, 1, 0, 0`, so gates 2 and 3 run the gear the other way). It also releases filament by moving to the opposing gate (`selector_release_gates`). You no longer need to change the load/unload directions by hand.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk