moggieuk/Happy-Hare #334: Bloblifier with Skew_correction

Dragi2k, 2024-07-06:
if you use a skew_correction it is possible that the Printer runs in "Move out of range" and the print starts with max part fan und no prime
Thats becouse Bloblifyer uses Max Y but with skew_correktion it is Possible that it tries to move to y+0.15 or something

thats why 
{% set position_y = printer.configfile.config["stepper_y"]["position_max"]|float %}
should be changed to 
{% set position_y = (printer.configfile.config["stepper_y"]["position_max"]|float-1) %}

moggieuk, 2024-07-08:
Thank you.  Agree this is an oversight.

@Dendrowen any change you can address?  I suspect there is a way to ask klipper for the max position less skew to minimize the reduction in y position (although that would imply telling klipper the z-height).  Maybe there is a max_skew variable?

Dendrowen, 2024-07-10:
To gather the skew it seems I'd need to overwrite the gcode macro and save the skew myself and somehow calculate the max y position. Possible of course, but it will take some time and tinkering. I could set something up. Would you be willing to test it, @Dragi2k? I don't have skew correction set up and haven't encountered issues so far, so won't be doing it.

Dragi2k, 2024-07-10:
yea i can test it
but would it not be simpler to say 
{% set position_y = (printer.configfile.config["stepper_y"]["position_max"]|float-{blob_skew_value} %}
and than create a value in the config 
example 1
i mean 1 mm from the back shoult be fine or not ? or is it possible to ask klipper if skew is enable and that on this case 1mm or 0,5mm from max y deducted

Dendrowen, 2024-07-16:
Might be a quick workaround. I've added this and pushed it to my personal fork so you can test it. There are some other small updates in it too. 
https://github.com/Dendrowen/Happy-Hare

```
cd ~
git clone https://github.com/Dendrowen/Happy-Hare Happy-Hare-Dendrowen
cd Happy-Hare-Denrowen
./install.sh
```

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

In v4 Blobifier has a y_offset and brush_y_offset setting, so you can keep its moves clear of the Y limit when skew correction is on.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk