moggieuk/Happy-Hare #337: Question about park/return park_vars.next_xy

weuzor, 2024-07-10:
Printing OrcaSlicer gcode containing assembled objects without prime tower, next pos x and next pos y are seemingly not correct.
The toolhead will return to near where it left the print (but surprisingly not at the same location), make a dot in the wrong color and then move to the position where the print is supposed to continue.
If the same is printed with unassembled objects the toolhead returns correctly to the position where the print continues.
Is that intended behavior or perhaps a bug? Should I use another slicer?

moggieuk, 2024-07-10:
That's interesting.  The Next X/Y move works by preprocessing and looking ahead for the next G1 move.  It's that simple.  To try to diagnose this we would need fragments of two gcode files with  100 lines or so around a particular toolchange.   One gcode with "assembled objects" and the other with "unassembled objects".  I don't user OrcaSlicer (maybe I should) but with the gcode samples what is happening should be obvious..

BTW HH v2.6.0 contains new logic to implement retraction while toolhead is being changed. This will also help with unexpected marking..

@Dendrowen flagging you on this one two.

Dendrowen, 2024-07-10:
This is strange indeed. I use orca slicer but haven't noticed this issue. I suppose it's an orca slicer 'bug' (between quotes, because orca slicer isn't really expecting the use of hh)

Maybe we're able to solve it but like @moggieuk said, we need fragments of both gcode files.

weuzor, 2024-07-10:
Hi! Thanks for the answers. Here are two zipped gcode files of the same print. 
[gcode_files.zip](https://github.com/user-attachments/files/16164306/gcode_files.zip)
multiple_objects.gcode  works fine the other one leads to described behavior.

weuzor, 2024-07-12:
In the meanwhile I found the true cause for wrong next_xy  to be the "Avoid crossing walls" option in Orca Slicer. Unchecking this option solves the problem.

closed 2024-11-27 by github-actions[bot]