moggieuk/Happy-Hare #338: Info on Why 'Timer Too Close' Can Happen - Public Service Bulletin

zeplintwo, 2024-07-10:
**Info on Why 'Timer Too Close' Can Happen - Public Service Bulletin**
This is not directly a bug.

When using the script from [Staubgeborener/klipper-backup](https://github.com/Staubgeborener/klipper-backup) to keep updated backups, the script can cause 'Timer too close' errors in conjunction with Happy-Hare and ERCF.

I experienced a major storm in my area, which left us without our primary ISP and its DNS routers. While running through 'personal hotspots'/cellphone ISP, the DNS to GitHub, even when not updating this script, would cause the Happy-Hare to trigger 'Timer too close' errors in Klipper.

Errata to Correct from 'FAQ':

- Set log_file_level: -1
- rPi temperature never exceeded: 52°C
- Edited ~/klipper/klippy/mcu.py by changing TRSYNC_TIMEOUT = 0.025 to TRSYNC_TIMEOUT = 0.05
- Removed 'moonraker-time-lapse' and associated 'webcam' plugins 'crowsnest'
- It wasn't until I fully uninstalled the CRON, rebooted, and removed the 'Gcode scripts' for backup, as well as all files related to the 'klipper-backup' plugin, that the system returned to stability.

Version Information:

- Happy-Hare: v2.5.2-37-g147f52a4
- Klipper: v0.12.0-158 (needs update)
- Mainsail: v2.10.0 (needs update)
- Moonraker: v0.8.0-379 (needs update)

Removed:

- Timelapse: v0.0.1-143-gc7fff11e
- Crowsnest: v4.1.8-1 (needed update)

The OS was updated to the current apt packages before the issue, but no specific versioning is available.

**Public Service Bulletin**
This is more of a public service bulletin indicating that this issue could happen in rare cases. It is not a bug that needs to be addressed by the Happy-Hare team. I do not think this is an issue with Klipper-backup directly. It seems more like a timer and DNS timeout bug causing an error to be generated by Klipper.

moggieuk, 2024-07-10:
Thanks. I can probably incorporate this into the Troubleshooting page of the wiki: https://github.com/moggieuk/Happy-Hare/wiki/Troubleshooting-and-Common-Issues

Note that another TTC source that was discovered (also related to too much file writing) is that the "save_variable" call in klipper is not very efficient.  HH stores a lot of state in it and thus can provoke TTC.  HH v2.6.0 (just make generally available) includes implementation of a batch update to `mmu_vars.cfg` that reduces the number of file write/read cycles from about 10 to 1 or 2.

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

Thanks for sharing this, it's a useful tip for anyone chasing "Timer too close" errors.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk