moggieuk/Happy-Hare #340: Endless Spool function change

platinumalloy01, 2024-07-11:
I have a suggestion for the endless spool function in happy hare on the ercf.
Rather than unload the filament once a runout has been detected, I recommend that it instead passes a specified amount of filament through the encoder/selector while continuing to print, then after that specified amount has passed through signifying that the gate is cleared and the selector is able to move to another gate loading the next roll of filament, and using a sensor less homing type function that detects when it pushes against the currently loaded filament then continues the print as normal, and not wasting filament, or pausing the print. Thank you for reading, I understand if this may be a limitation of the host computer, but maybe there’s a way.

moggieuk, 2024-07-14:
This has been asked quite a few times before.  The problem is that it could work (see not super easy) ONLY with single color endless spool prints...  once the filament break is past the MMU gear, there is no way to color change. Also, if the print happens to end with the break still in the bowden it also becomes a hassle.

Frankly, loosing 600mm - 1.5m of filament seems like a better compromise than the extra complexity trying to special case this would bring.  That said, the application of this idea to single color prints is on my "might do" list but near the bottom :-)

platinumalloy01, 2024-07-14:
Thank you for responding, I figured it might not be feasible at this time, otherwise it would have been implemented already. Only reason I even thought about it was because of an issue I came across where the filament was unable to backfeed because it was getting trapped in the filament block at the buffer system, ended up being the Bowden but yeah, anyways issue resolved. Thanks again for the response and explanation.

moggieuk, 2024-07-15:
You are right about the trapping in the buffer system .. some are more prone than others. Filamentalist will work quite reliably I think.  The default ERCT can have issues.   I know KinematicDigit is working on some improvements.  Also, the v2.6.0 of HH includes a designated "waste gate" option so one gate can be used to dump filament fragment.  This is experimental and suggested by KinematicDigit, but the downside is that the pre-gate sensors are ignored for runout because the filament must pass clear through the selector so it can select the waste gate...

k1-801, 2024-08-14:
I know this was already marked as "wontfix", but I think there is a way to improve the EndlessSpool feature for the scenarios where it would be possible to continue printing with what is left in the filament channel.

The idea is to extend the gcode preprocessing in order to add a parameter to __every__ Tx macro that would specify the length of filament that will be extruded in the specified toolchange (i.e., before the next Tx macro or before the end of file). After the toolchange, we reset the counter to whatever length was specified, and, if we encounter a runout, we compare what is left with the length of our bowden tube.

From here, we have several different scenarios
* The remaining length in the toolchange is greater than the length of bowden tube. It means, there is no reason to unload the filament, and we can simply continue printing. If there is an extruder entry sensor, we can keep printing all the way until we hit that sensor, then we have to load the next spool. Otherwise, we need to follow the remaining bit of filament with the next spool for this tool since the moment we encounter a runout, in a synchronized manner. Ideally, the selector would move to the next spool and start feeding it wiithout pausing the print. The option with no extruder entry sensor should be available regardless of the actual sensor's presence, as such sensor may be too close to the extruder gears, and an arc gcode command could take too long
* The remaining length in the toolchange is *slightly* shorter than the length of bowden tube (by a configurable amount, 10cm or so). It means, we won't have to follow it with the next spool, but there is no reason to stop the print, we can just eject the remaining bit through the nozzle. Optionally, we can ask for the user to just remove the remaining bit manually (should be configurable).
* The remaining length in the toolchange is *significantly* shorter than the length of bowden tube. In this case, we unload as we currently do.

If the user wishes to continue printing even if the bit will not be able to be unloaded automatically, they can specify the "extruder purge" length to be very long and mark it for manual removal.

Optionally, we can remember that a certain gate still has a certain length of filament mapped to a certain tool unloaded to it, and if we encounter a Tx macro with the same tool and a length greater than that, we can try to "use up" that piece. That would, however, require introducing additional statuses to gates ("R" for "Running out").

platinumalloy01, 2024-08-14:
Thanks man, this is awesome, hope it can be implemented.

ac1dwalk3r, 2024-09-21:
i also opened an issue about that a few months ago....i thought about that a while....first of all, pushing back an empty spool is most often not possible in many MMU's cause the end basically has always a kink....so having a eject gate definitly doesn't work....atm i'm testing pregate sensors with a ballbearing, we'll see how the eject works with them....hopefully the bowden straightens it enough to get out....k1-801 comment is quite interesting, also thought about the gcode preprocessing, but i'd say that's something for self implementation in some costumized macros.....the easiest one being if Toolchanges are less than 2, enable Endless Spool without Eject....but still this function needs to be implemented into HappyHare first, but moggieuk doesn't seem to be interested......anyway, Endless Spool is one of the more problematic features in this Software

k1-801, 2024-10-30:
I began implementing the pre-processing routine for gathering filament lengths that are to be used between toolchanges.
However, the current implementation "buffers" the part of the file between the `T?` macro and the next `G1` move macro. The way suggested pre-processing of extruding moves is meant to be implemented, it would require to buffer everything between the `T?` macros, which may end up being multiple megabytes, if not the entire file. To circumvent this, I'm gonna have to rewrite the pre-processing in a non-buffering way. It may cause issues if the file is somehow "streamed" to the pre-processor instead of being read from disk.
Additionaly, I am having some trouble with `LOAD_INITIAL_TOOL`, as it behaves the same way as a `T?` macro, and, thus, requires a length specified.
I will figure it all out eventually and I will present my changes in a few days for review. It will not be a functioning implementation of an EndlessSpool, just the pre-processing for now. However, I hope that the data of expected extrusion length will come in handy for other uses as well.

k1-801, 2024-10-31:
@moggieuk if you have some free time, I would like to hear your thoughts regarding this proposal.
I made a mock-up of the proposed pre-processing and posted it to #491, though it is yet to be tested (didn't even check if it's valid python code yet :D)
Some math operations probably need type conversions or maybe there are a couple of logical errors, I will test it as soon as I have some time for that myself.

The important part here is: with this new preprocessor, we know how much filament is meant to be extruded between the tool changes, and we know how much of that has already been extruded at the point where runout logic was triggered, therefore, we know how much more we need before the toolchange - and we know (approximately) how much is available in the bowden.

k1-801, 2024-10-31:
Debug in progress

[image]

k1-801, 2024-10-31:
[image]
Other than the fact that it duplicates the lines, I think it did a pretty good job

k1-801, 2024-10-31:
[image]
Duplication fixed.

moggieuk, 2024-11-02:
Unfortunately I'll say up front that I have very limited time.   I do find this idea very interesting but I see a host of issues even after the pre-processing is done so I'll offer comments on those because clearly you have the moonraker parser under control (aside from parsing time :-)

1. Happy Hare will need to track filament extruded since last toolchange -- easy
2. On runout, HH needs to know the amount of filament from runout sensor to nozzle.  This is an awkward calibration...
3. With HH v3.0.0 you can optionally have individual post-gate sensors (like Box Turtle design) also you can have variable bowden lenghts (one per gate).
3a. Therefore possible runout starting points are: individual mmu_pre_gate sensor, shared mmu_gate sensor, individual post_gate sensors, extruder_entry sensor, (I guess toolhead sensor but I don't currently monitor that one) and the encoder.
3b. The encoder position is variable and dictated by the  current "calibrated_clog_detection_length"
3c. All starting position are subject to the moves that happen from the moment of sensor detection to the point at which the print stops ... generally there are about 5 moves queued up.  Short extrusion moves aren't a problem but long ones can move the effective position quite a bit.  This distance really needs to be captured to get an accurate starting point.
4. Adding the remaining filament to the purge volume .. that is likely to be massive if a color change is pending soon after the point of no return.  There is a new single point of calculating the purge volume and supplying as a printer variable already `printer.mmu.toolchange_purge_volume` .. would need to be honored by Blobifier or other purge mechanisms.

All of this isn't unsurmountable but want to point out what's left in store...

closed 2024-12-16 by github-actions[bot]