moggieuk/Happy-Hare #344: Error restoring position after MMU Pause

JeeAreOh, 2024-07-18:
I was printing a tall piece when HH detected a run out. I've solved the run out issue, and then I resumed the printing using the HH Kliperscreen interface.
The toolhead moved slowly to the piece, restoring X, Y and Z position, all at once. Unfortunately, restoring the position this way, the toolhead hit the piece, and ripped it off the bed.
Is there a way to restore Z position completely before restoring the other two (X and Y)?
I guess that it would be a great solution.

Thank you @moggieuk 
You're doing a great job maintaining Happy Hare! ✌️

moggieuk, 2024-07-18:
Yikes. Definitely should not happen.  Can you send the `mmu.log`.   I could be a recent change on the introduction of `z_hop_ramp` but I need to logs to help track it down.

JeeAreOh, 2024-07-18:
@moggieuk thanks for the fast reply.
I've reproduced the same error and downloaded the mmu.log file. Here it is:

[mmu.log](https://github.com/user-attachments/files/16283764/mmu.log)

zenduin, 2024-07-24:
I'm having the same issue. Looks like it did restore Z but then went back down to park/brush (mine are the same) position, then over to the bucket, then straight back to the print, which for me just dislodged my prime tower, but that is still enough to ruin the print.
Also running Voron2.4 with Blobifier, but with Tradrack 1.0e and no filament cutting.
[mmu(3).log](https://github.com/user-attachments/files/16367710/mmu.3.log)

zenduin, 2024-07-29:
Adding my log from testing Blocked Filament Path recovery since the tool crashed into the prime tower because of this issue.
I think the only problem is the tool is at the Blobifier park position when any problem occurs, so after calling RESUME and restoring Z, the tool head will travel back to the Blobifier park position instead of the position where the tool change was called.
[mmu(7).log](https://github.com/user-attachments/files/16420104/mmu.7.log)

moggieuk, 2024-07-30:
What you describe (returning to the blobifer park position could be what's happening.  I'm going to cc @Dendrowen the author of Blobifer to see if he has a hunch.

One thing you can do (and is recommended with blobifer is to set:
```
variable_restore_xy_pos         : "next"        ; last|next|none - What x,y position the toolhead should travel to after a toolchange
```
This will restore the toolhead to the next print location and NOT where the pause happened.

I still wonder if there is an issue, but try the above suggestion as a temp workaround.

JeeAreOh, 2024-07-30:
> What you describe (returning to the blobifer park position could be what's happening.  I'm going to cc @Dendrowen the author of Blobifer to see if he has a hunch.
>
>
>
> One thing you can do (and is [...]

Ok @moggieuk I'll try this solution and I'll keep you informed. Thx

JeeAreOh, 2024-07-30:
> What you describe (returning to the blobifer park position could be what's happening.  I'm going to cc @Dendrowen the author of Blobifer to see if he has a hunch.
>
>
>
> One thing you can do (and is [...]

Actually, my variable is already in "next". Unfortunately, it wasn't a solution.
As a workaround, I'm increasing Z toolhead position higher than the part I'm printing, then I resume. The issue is not solved, but at least the toolhead don't hit the part during the position restoration.

JeeAreOh, 2024-07-30:
Maybe @Dendrowen  could've help us somehow.
Thank you in advance.

moggieuk, 2024-07-30:
Have you verified, by look at your gcode that the "next" pre-processing has been successful.  All `Tx` toolchange commands should have been changed to `MMU_CHANGE_TOOL TOOL=x NEXT_POS=x,y`

Let's start there..

zenduin, 2024-07-30:
Using the Happy Hare client macros fixed the problem for me. I was using the built-in Klippain pause/resume/cancel macros. Now the toolhead keeps the nozzle on the Blobifier tray during pause, but it's an improvement over crashing on resume.
Looks like my "next" pre-processing isn't working though, I'm still seeing just T0 and T1 in this file, but it was processed and I can see things like !purge_volumes! have been filled in.

moggieuk, 2024-08-01:
Progress...  I will talk to the Klippain folks.

The "next" pre-processing. Make sure these are set in `moonraker.conf`:
```
[mmu_server]
enable_file_preprocessor: True
enable_toolchange_next_pos: True
```

Also, I recommend adding this (its new so need the latest moonraker):
```
[file_manager]
default_metadata_parser_timeout: 30
```

zenduin, 2024-08-01:
[mmu_server] was already set, adding the [file_manager] section had no effect.
I'm seriously considering just removing Klippain at this point, seems it causes me more pain.

I'm using a Raspberry Pi 4B 2GB & MainsailOS (Debian Bullseye) if that helps. 
I'm also considering wiping the SD card and starting over using KIAUH.

JeeAreOh, 2024-09-01:
> Have you verified, by look at your gcode that the "next" pre-processing has been successful.  All `Tx` toolchange commands should have been changed to `MMU_CHANGE_TOOL TOOL=x NEXT_POS=x,y`
> 
> 
> 
> Also, are you using the Happy Hare client macros for pause/resume/cancel?
> 
> 
> 
> Let's start there..
> 
> 

Using the client macros solved the problem. Thank you guys for helping me.

closed 2024-09-02 by JeeAreOh