moggieuk/Happy-Hare #345: Over extruding and resetting Klipper

Northernboy, 2024-07-21:
I am trying to get my ERFC V2 working on my RatRig V-Core 3 printer using Happy Hare. But when the printing starts the extruder goes too fast and eventually cause my tool board to reset n stopping the print. This will only last a couple of seconds. I tried reducing the max extrude cross section, but it will stop as well cause requested it greater than this. Went through my setting and they seem fine. If I disable the MMU it will print fine with a single Color. Any help with this would be greatly appreciated.

Thx
NB

Northernboy, 2024-07-22:
Are you using the same printer as me? Also which slicer program are you using?

moggieuk, 2024-07-24:
I don't know much about the extruder setup on the RatRig. Can you share the klipper configuration for the toolhead?

Also, can you run this can report on what it says (hidden debugging command):
```
_MMU_DUMP_TOOLHEAD
```

Secondly, could you try adding this to your `mmu_parameters.cfg` file:
```
homing_extruder: 0
```
It is a hidden option but changes the way the extruder is controlled and it might help pinpoint the problem.

If still no luck please turn the most verbose logging option in `mmu_parameters.cfg`
```
log_file_level: 4
```
and attach the `mmu.log` file (same directory as klippy's logs)

Northernboy, 2024-07-24:
Thanks for the information, I’m going to try SuperSlicer instead of Simplify3D first. Then I’ll try your pointers and submit more info. My extruder is Rapido UHF with Orbiter V2.

Northernboy, 2024-07-27:
It’s actually working using SuperSlicer!!!
I have some settings to tweak but atleast it’s printing. I guess Simplify3d isn’t meant for MMU.

moggieuk, 2024-08-01:
@Northernboy @salocain can you help me by sending a (smaller) gcode file that is produced by Simplify3D...

It is true that testing only is done on PrusaSliver, SuperSlicer and the new favorite Orca Slicer, but this is no reason why Simplify3D can't be supported - just need some reference gcode files to see what it is producing..  firmware retraction, etc..?

Northernboy, 2024-08-05:
I’ll give Simplify3D another try in the near future just been working with SuperSlicer and starting to get use to it. My ERCF seems to be working fairly good just have a few adjustments to make.

closed 2024-11-13 by github-actions[bot]