moggieuk/Happy-Hare #347: GDW DS041MG Servo

alidaf, 2024-07-23:
The SIboor kits come with a DS041MG servo but this doesn't work as expected (very small movements) until the mmu_servo.ppy file is modified. Siboor posted a fix on Discord that involves using sed to remove line 83 vis:

`sed -i '83d' ~/Happy-Hare/extras/mmu_servo.py`

This removes the following problematic line:

`self.mcu_pwm.set_pwm(end_time, 0.)`

The last update restored the original mmu_servo.py file.

This seems like a hack so perhaps there is a better and permanent method.

luismello, 2024-07-23:
Hi

Activate this below in mmu_parameters.conf

servo_always_active: 1 			# CAUTION: 1=Force servo to always stay active, 0=Release after movement

that solved the problem for me

moggieuk, 2024-07-24:
Yes, the Siboor "hack" was a temp solution and very unreliable as HH evolves. The option mentioned above `servo_always_active: 1` is my permanent fix.

alidaf, 2024-07-25:
Many thanks to you both.

closed 2024-07-25 by alidaf