moggieuk/Happy-Hare #348: Tradrack Load Unload Delta (Encoder & Gate Endstop)

ningpj, 2024-07-23:
OK, so I think I have tracked down the delta between reported unload / load distances for Tradrack with Encoder, Gate end stop and toolhead sensor. On load, HH doesn't appear to account for the distance from where the filament is parked > gate sensor > encoder (45mm on my setup).  On unload it doesn't appear to account for encoder to gate sensor (28mm on my setup).

This is cosmetic and doesn't affect the operation of HH.

```
gate_parking_distance: 17
gate_endstop_to_encoder: 28
Delta before encoder notices load: 45
mmu_calibration_bowden_length: 1115.9
```

Spreadsheet with load and unload traces and calculated / adjusted distances attached for reference.
[TR with Encoder Gate Sensor and Toolhead Sensor Load Unload Delta.xlsx](https://github.com/user-attachments/files/16347738/TR.with.Encoder.Gate.Sensor.and.Toolhead.Sensor.Load.Unload.Delta.xlsx)
.

moggieuk, 2024-07-24:
Thanks!   I'll try to get this into the upcoming v2.7.0 release cosmetics are important (at least to me) :-)

moggieuk, 2024-07-30:
@ningpj have a few questions.  When "loading the gate" - the very first step

```
    # Load filament into gate. This is considered the "zero" positon for filament loading. Note that this
    # may overshoot for the "encoder" loading technique but subsequent bowden move will accommodate. Also
    # for systems with gate sensor and encoder with gate sensor first, there will be a gap in encoder readings
    # on next bowden move
    def _load_gate(self, allow_retry=True, adjust_servo_on_error=True):
    ```
    This is the comment my code to serve as a reminder to me.
    
    For the encoder case, the filament position is initialized to:
    ```
    self._set_filament_position(measured)
    ```
    where measured is the overshoot from when the encoder first moves.
    
    For the gate switch case, after homing to the switch, that is considered the 0mm point:
    ```
    self._initialize_filament_position()
    ```
    It was kind of an arbitrary zero point and I guess it would be more strictly accurate to be the gate parking position which would be the same in both cases...
    
    As you say this is cosmetic but it would be nice for the encoder to match

moggieuk, 2024-07-31:
Actually no need to test.  I've fixed this and is wasn't only for Tradrack.  For historical reasons I has the 0mm point at the start of the fast bowden move.  I have move this to the gate parking position for consistent readings on load and unload.
Fixed in v2.7.0 release

closed 2024-09-09 by ningpj