moggieuk/Happy-Hare #349: No slicer tool map loaded

zenduin, 2024-07-24:
I'm trying to get Blobifier to work with different purge volumes defined by the slicer but it always reverts to default purge volume.
MMU_SLICER_TOOL_MAP returns the error: "No slicer tool map loaded".

Here is my pre-processed start gcode:
>MMU_START_SETUP INITIAL_TOOL=3 REFERENCED_TOOLS=0,1,2,3 TOOL_COLORS=E2DEDB,080A0D,446B20,E0191E,F3B400,66FF00,FFD700,002364,FF00FF TOOL_TEMPS=215,215,215,215,215,215,215,215,215 TOOL_MATERIALS=PLA,PLA,PLA,PLA,PLA,PLA,PLA,PLA,PLA PURGE_VOLUMES=0,70,92.5,197.5,220,227.5,222.5,135,240,497.5,0,282.5,365,565,575,612.5,160,440,395,60,0,165,395,385,442.5,100,325,475,87.5,202.5,0,382.5,482.5,447.5,135,262.5,325,115,152.5,175,0,205,122.5,160,237.5,335,120,155,225,195,0,242.5,162.5,240,280,125,160,205,60,180,0,167.5,245,510,60,290,382.5,570,575,610,0,402.5,482.5,100,142.5,205,442.5,447.5,487.5,147.5,0
MMU_START_CHECK
START_PRINT EXTRUDER_TEMP=215 BED_TEMP=60 MATERIAL=PLA SIZE=196.225_165.032_262.625_303.111 INITIAL_TOOL=3 TOOLS_USED=0,1,2,3 CHAMBER=0
MMU_START_LOAD_INITIAL_TOOL

It looks like MMU_START_SETUP calls MMU_SLICER_TOOL_MAP, but during the print it is not loaded.
From my MMU_Log:

> 18:28:38 > MMU_SLICER_TOOL_MAP RESET=1 INITIAL_TOOL=3 PURGE_VOLUMES=0,70,92.5,197.5,220,227.5,222.5,135,240,497.5,0,282.5,365,565,575,612.5,160,440,395,60,0,165,395,385,442.5,100,325,475,87.5,202.5,0,382.5,482.5,447.5,135,262.5,325,115,152.5,175,0,205,122.5,160,237.5,335,120,155,225,195,0,242.5,162.5,240,280,125,160,205,60,180,0,167.5,245,510,60,290,382.5,570,575,610,0,402.5,482.5,100,142.5,205,442.5,447.5,487.5,147.5,0
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=0 TEMP=215 MATERIAL=PLA COLOR=E2DEDB  QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=1 TEMP=215 MATERIAL=PLA COLOR=080A0D  QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=2 TEMP=215 MATERIAL=PLA COLOR=446B20  QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=3 TEMP=215 MATERIAL=PLA COLOR=E0191E  QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=4 TEMP=215 MATERIAL=PLA COLOR=F3B400 USED=0 QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=5 TEMP=215 MATERIAL=PLA COLOR=66FF00 USED=0 QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=6 TEMP=215 MATERIAL=PLA COLOR=FFD700 USED=0 QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=7 TEMP=215 MATERIAL=PLA COLOR=002364 USED=0 QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP TOOL=8 TEMP=215 MATERIAL=PLA COLOR=FF00FF USED=0 QUIET=1
18:28:38 > MMU_SLICER_TOOL_MAP
18:28:38 --------- Slicer MMU Tool Summary ---------
         4 color print (Purge volume map loaded)
         T0 (Gate 0, PLA, e2dedb, 215°C)
         T1 (Gate 1, PLA, 080a0d, 215°C)
         T2 (Gate 2, PLA, 446b20, 215°C)
         T3 (Gate 3, PLA, e0191e, 215°C)
         Initial Tool: T3
         -------------------------------------------
         DETAIL=1 to see purge volumes

I also don't see the tool colors reflected on Mainsail.
I am able to use the MMU_SLICER_TOOL_MAP command to set the purge volumes myself using the console.
[mmu(3).log](https://github.com/user-attachments/files/16367834/mmu.3.log)

moggieuk, 2024-07-25:
MMU_SLICER_TOOL_MAP will only display the map once it is loaded.  This is generated by the command:
```
18:28:38 --------- Slicer MMU Tool Summary ---------
4 color print (Purge volume map loaded)
T0 (Gate 0, PLA, e2dedb, 215°C)
T1 (Gate 1, PLA, 080a0d, 215°C)
T2 (Gate 2, PLA, 446b20, 215°C)
T3 (Gate 3, PLA, e0191e, 215°C)
Initial Tool: T3
-------------------------------------------
DETAIL=1 to see purge volumes
```
so it is loaded.

If you `mmu_pause` the print. Then run:
```
MMU_SLICER_TOOL_MAP DETAIL=1
```
you should see the purge volume map.

The issue with colors not showing in Mailsail is fixed with a forced browser refresh (shift-reload page) as explained in the doc.  This is fixed in the upcoming v2.7.0 release, but until then the manual UI refresh is necessary

demitrix, 2024-07-27:
I am experiencing similar behavior. Trying to troubleshoot another issue and noticed this. Even just running
```
MMU_START_SETUP INITIAL_TOOL=1 REFERENCED_TOOLS=1 TOOL_COLORS=000000,000000,000000,0000FF,00FFFF,C0C0C0,FFFFFF TOOL_TEMPS=230,260,230,230,230,230,230 TOOL_MATERIALS=PLA,ASA,PLA,PLA,PLA,PLA,PLA PURGE_VOLUMES=0,60,60,307,586,459,560,60,0,60,307,586,459,560,60,60,0,307,586,459,560,80,80,80,0,540,562,661,114,114,114,256,0,272,408,60,60,60,241,213,0,210,80,80,80,272,242,60,0
MMU_START_CHECK
; START_PRINT TOOL_TEMP=260 BED_TEMP=105 FILAMENT_TYPE=ASA
MMU_SLICER_TOOL_MAP
```
Results in the last command reporting "No slicer tool map loaded"
<details><summary>mmu.log</summary>

```
21:03:22 > MMU_SLICER_TOOL_MAP RESET=1 INITIAL_TOOL=1 PURGE_VOLUMES=0,60,60,307,586,459,560,60,0,60,307,586,459,560,60,60,0,307,586,459,560,80,80,80,0,540,562,661,114,114,114,256,0,272,408,60,60,60,241,213,0,210,80,80,80,272,242,60,0
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=0 TEMP=230 MATERIAL=PLA COLOR=000000 USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=1 TEMP=260 MATERIAL=ASA COLOR=000000  QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=2 TEMP=230 MATERIAL=PLA COLOR=000000 USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=3 TEMP=230 MATERIAL=PLA COLOR=0000FF USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=4 TEMP=230 MATERIAL=PLA COLOR=00FFFF USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=5 TEMP=230 MATERIAL=PLA COLOR=C0C0C0 USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP TOOL=6 TEMP=230 MATERIAL=PLA COLOR=FFFFFF USED=0 QUIET=1
21:03:22 > MMU_SLICER_TOOL_MAP
21:03:22 --------- Slicer MMU Tool Summary ---------
         Single color print
         T1 (Gate 1, ASA, 000000, 260°C)
         Initial Tool: T1
         -------------------------------------------
         DETAIL=1 to see purge volumes
21:03:22 > MMU_CHECK_GATE TOOLS=1
21:03:22 > _MMU_PRINT_START
21:03:22   DEBUG: Job State: CANCELLED -> STARTED (MMU State: Encoder: Enabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
21:03:22   DEBUG: Setting servo to down (filament drive) position at angle: 1
21:03:22   DEBUG: Syncing gear stepper and extruder
21:03:22 Modifying MMU gear stepper run current to 70% for extruder syncing
21:03:23 Happy Hare initialized ready for print (no filament preloaded)
21:03:23   DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: True, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
21:03:23   DEBUG: Setting servo to move (filament hold) position at angle: 145
21:03:24 Checking Gate 1...
21:03:24   DEBUG: Setting servo to down (filament drive) position at angle: 1
21:03:24   DEBUG: Unsyncing gear stepper and extruder
21:03:24 Restoring MMU gear stepper run current to 100% configured
21:03:25   DEBUG: Gate endstop reached after 15.2mm (measured 0.0mm)
21:03:25 Tool T1 - Filament detected. Gate 1 marked available
21:03:27   DEBUG: Selecting tool T0 on Gate 0...
21:03:27   DEBUG: Setting servo to move (filament hold) position at angle: 145
21:03:28   DEBUG: Setting servo to up (filament released) position at angle: 145
21:03:28 Tool T0 enabled
21:03:28 Gates: | 0 | 1 | 2 | 3 | 4 | 5 | 6 |
         Tools: |T0 |T1 |T2 |T3 |T4 |T5 |T6 |
         Avail: | S | S |   |   |   |   |   |
         Selct: | * |------------------------ T0
21:03:28 > __MMU_GATE_INSERT
21:03:28 > MMU_SLICER_TOOL_MAP
21:03:28 No slicer tool map loaded
21:03:28 > MMU_PAUSE MSG="Load bypass or initial tool before resuming print"
21:03:28 MMU issue detected. Print will be paused
         Reason: Load bypass or initial tool before resuming print
21:03:28 After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
21:03:28   DEBUG: Cannot save toolhead position or z-hop for mmu_pause because not homed
21:03:28   DEBUG: Job State: PRINTING -> PAUSE_LOCKED (MMU State: Encoder: Disabled, Synced: False, Paused temp: 0.0, Resume to state: printing, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 72000.00s)
21:03:28 > PAUSE
21:03:28   DEBUG: Cannot save toolhead position or z-hop for pause because not homed
21:03:28 Attempting to recover filament position...
21:03:28   DEBUG: Checking for filament in MMU...
21:03:28   DEBUG: Filament not detected by sensors: toolhead, mmu_gate, extruder
21:03:28 [T0] < (g) .. En ......... (e) .. [Ex .. (t) .. Nz] UNLOADED -18.2mm (e:0.0mm)
```

</details>

moggieuk, 2024-07-28:
i pushed a fix for this to the main branch...

It was a race condition that could occur when "octoprint support" was off.  Anyway, Happy Hare detected the print started AFTER the slicer tool map was loaded.  On the print start it cleared the map.

Let me know if latest patch fixes the issue.

zenduin, 2024-07-29:
Looks to be working perfectly now!

closed 2024-07-29 by zenduin