moggieuk/Happy-Hare #350: Blocked Filament Path Recovery

zenduin, 2024-07-24:
During a print when the gate is blocked by filament, Happy Hare detects the problem and attempts to recover, however the servo does not move into the down position before trying to retract filament. So the gear stepper spins without being engaged and then reports that recovery has failed (big surprise).
I've tried to reproduce this behavior outside of a print, but recovery succeeds.
Attaching the same log on my third post today (yes, three issues in one print).
[mmu(3).log](https://github.com/user-attachments/files/16367911/mmu.3.log)

Voron 2.4 with Blobifier & Tradrack 1.0e

moggieuk, 2024-07-25:
That used to work so must be a regression.  I will put on the list to retest..

moggieuk, 2024-07-25:
I think I see the problem... you have both an encoder AND gate filament sensor I presume?   If see a logic flaw and doesn't issue servo down...

moggieuk, 2024-07-25:
[image]

If you are brave and don't want to wait for me, carefully add this green line to your `Happy-Hare/extras/mmu.py` file.  Mainsail will show your install as "dirty" but I'll get this in to the next update.

moggieuk, 2024-07-28:
I've patched the main branch now so an update should fix this.

zenduin, 2024-07-29:
Okay, servo now moves into the down position correctly, and it does manage to recover sometimes.
However, the filament is still being pinched between the selector cart and the gate so it can't actually be pulled out by the gear motor.
Maybe make the selector move slightly in the opposite direction so it stops pinching the filament?

I managed to test this three times by pushing filament back through the gate, twice failed on T1, one success on T0, and then my toolhead crashed into the prime tower due to #344.

ningpj, 2024-07-29:
If you havent, you can manually calibrate your gate offsets to make sure every gate is perfectly aligned with the selector.  If you have already done this try redoing the calibration of gates pinching filament.   

https://github.com/moggieuk/Happy-Hare/wiki/MMU-Calibration#---step-1-calibrate-selector-offsets

moggieuk, 2024-08-07:
If the filament is being pinched (sideways force) then it is a mechanical problem.  Either slightly misaligned gate or perhaps a bulge at the end of the filament caused by previous tip forming..  Also is the servo moving to fully locked position? If so there should be plenty of grip..

zenduin, 2024-08-07:
Gate is aligned and filament is not pinched until the selector tries to move to a different gate. The selector moving is what is causing the pinching, so I think it would be a good idea to move the other way just a bit so it isn't pinching before trying to retract filament again. Even just turning off the selector motor would help.
This probably isn't a problem on the ERCF because the gates and selector cart have a fillet around the filament hole, but on Tradrack the holes are straight.
Also, the filament brake in ERCF is pretty useless, and really just makes it more of a pain to load filament. I tried bother versions when I was still screwing with the ERCF.

closed 2024-11-13 by github-actions[bot]