moggieuk/Happy-Hare #353: how to downgrade?

salocain, 2024-07-25:
hello , with my problem on the 2.6 version , i looking for return to 2.5 for print .

thank you

salocain, 2024-07-28:
any reponses ? because i'm in an closed issue for now , and the downgrade could be a good option :-)

moggieuk, 2024-07-28:
you can pull any version from github and would then need to put back your old configuration files from the backup created in:
```
printer_data/config/mmu-2024*****
```
This is no doc on how to do this.   I will say that v2.6.0 has the same core functionality and should work the same as previous.  Note that I can ONLY support the latest revision .. there is only one of me :-)

salocain, 2024-07-28:
strange i do that with my 2.5 backup and same problems happened .

discord is not my preference

moggieuk, 2024-07-30:
You commented out:
`#extruder_homing_current: 40`

This would be replaced on upgrade.  Change this to:
`extruder_homing_current: 100`

Which is the same effect as commenting it out.

salocain, 2024-07-30:
i deleted HH ,config etc, klipper , and reinstall all --> i have always the over extrusion problem with happy hare , over extrusion is little word , because all extusion is x100 (really)

moggieuk, 2024-08-01:
over extrusion while printing?  If extrusion multiplier is set to 100 then HH does not / cannot change the extruder movement

moggieuk, 2024-08-01:
[image]
Here is example of a print when set up correctly.  Not a hint of over-extrusion..

closed 2024-08-01 by moggieuk

salocain, 2024-08-01:
thanks for your pick but , i do the same things before update , but after update , the extrusion is completly strange !!!

can you look at my first issue post [https://github.com/moggieuk/Happy-Hare/issues/343](url)

i'm always looking for an explaination and solution