moggieuk/Happy-Hare #355: HH KlipperScreen-start.sh respawning repeatedly

Sparkss, 2024-07-27:
I checked a printer running "stock" KS at the same version and do not see this behavior, only on the printer running HH KS (which is awesome, btw the way). Syslog, journalctl are flooded with restarts occurring every 4 ~ 5 seconds.

happy-hare
v2.6.0-52-g2ee2a39c
klipper
v0.12.0-267-g12cd1d9e
KlipperScreen
v2.2.2-417-g2f8bf94

Here is the service Info, which shows the last couple of restarts.

 service KlipperScreen status
* KlipperScreen.service - KlipperScreen
     Loaded: loaded (/etc/systemd/system/KlipperScreen.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2024-07-27 07:48:03 EDT; 1h 59min ago
   Main PID: 864 (KlipperScreen-s)
      Tasks: 7 (limit: 1387)
        CPU: 5min 24.358s
     CGroup: /system.slice/KlipperScreen.service
             |- 864 /bin/bash /home/pi/KlipperScreen/scripts/KlipperScreen-start.sh
             |- 884 /usr/bin/xinit /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py
             |- 889 /usr/lib/xorg/Xorg :0
             `-1145 /home/pi/.KlipperScreen-env/bin/python /home/pi/KlipperScreen/screen.py

Jul 27 09:47:25 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:25,978 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:26 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:26,016 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:26 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:26,052 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:30 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:30,957 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:31 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:30,999 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:31 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:31,037 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:35 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:35,936 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:35 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:35,963 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:35 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:35,989 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>
Jul 27 09:47:40 doomv2 KlipperScreen-start.sh[1145]: 20240727 09:47:40,965 [printer.py:init_temp_store] - Temp store: ['heater_bed', 'temperature_fan rpi_fan', 'temperature_sensor Raspberry_Pi', 'temperature_se>

When I run the command interactively I get the following error:

:~/KlipperScreen/scripts $ ./KlipperScreen-start.sh
Running KlipperScreen on X in display :0 by default

(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)
^[X connection to :0 broken (explicit kill or server shutdown).

I will add that KS is running just fine, which explains the .X0-lock file, but I am not sure why it keeps trying to respawn. I looked in the systemd/system folder and there was only 1 KlipperService file. I also only saw one instance running in the process list. Was there a step I may have missed in setting up HH KS?

Nothing shows up in the klippy.log, so nothing to attach.

moggieuk, 2024-07-28:
Missfield issue.  This for Happy Hare, not KlipperScreen.

Looking at the error, this does look like a system configuration issue or other KlipperScreen setup.   I would revert back to the default KS (which no doubt would have the same problem) and ask for support on the KlipperScreen forum or file an issue here: 
https://github.com/KlipperScreen/KlipperScreen

Sparkss, 2024-07-28:
You are correct, my apologies. I found there is already an issue opened with KS for it here: https://github.com/KlipperScreen/KlipperScreen/issues/1441

closed 2024-07-28 by Sparkss