moggieuk/Happy-Hare #356: How do I get the purge blob to stop right after tool change, and before it starts moving on purge tower?

heissenberger, 2024-07-28:
So after every successful filament change the tool head proceeds to purge tower and the nozzle lowers and it purges a big blob and then starts to move on purge tower.   I’m using prusa slicer and can’t get this to stop.

heissenberger, 2024-07-28:
Also.  Don’t know if this is the right place to ask but I can’t figure this out and I’m so damn close having this ERCF to work…..    any help would be great.

moggieuk, 2024-07-28:
Most likely this is a slicer config issue where it is introducing additional gcode commands. Read here: https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup

Or the calibration of your toolhead dimensions isn't correct. Read here: https://github.com/moggieuk/Happy-Hare/wiki/Blobbing-and-Stringing

v2.6.0 introduces a `toolchange_retract` setting that will completely eliminate any blobbing once the other dimensions are correct.  Here is a user example...

[image]

closed 2024-11-13 by github-actions[bot]