moggieuk/Happy-Hare #357: Different behavior when parking filament on ERCF

protorayn, 2024-07-28:
I can't for the life of me figure out why parking filament after checking gates works just fine but when ejecting filament form the bowden tube, whether on a print or by ejecting using KS Happy hare, the filament is ejected past the gears. 

I ran a few tests to check parking distance at the gate and when running the check gate macro it parks just fine. Any change I make to the parking distance variable reflected physically. After making sure the filament is parked flush with the gate I load the filament and extrude a small amount to make sure the filament is in the nozzle. I run the eject from KlipperScreen and it parks the filament so far back that the ERCF can no longer grip the filament.

I'm at a loss as I don't see any other variables that could be affecting the parking distance that is chosen. I don't see any physical reason for the filament to park normally using one method and not during print or long bowden move. 

Could the encoder delta be affecting the parking routine? It seems logical to me that the parking routine should be looking for the encoder to stop and then count steps to reverse out of the selector. 

[klippy.log](https://github.com/user-attachments/files/16405336/klippy.log)
[mmu.log](https://github.com/user-attachments/files/16405337/mmu.log)

moggieuk, 2024-07-30:
Ok. The problem isn't anything to do with gate parking distance - I would reset that back to default (it really shouldn't be necessary to change).  Instead look at the extruder unloading .. it is going too far and thus the "dumb" fast bowden unload goes too far.

I suspect that:
1) possibly your have the tip cutting macro set up to do a "final_eject" for something isn't configured correctly on the extruder unload.
2) The other possibility is that your calibrated bowden length is too long.  Would cause the filament to ram into the extruder gear on load (not the end of the world) and run past the gate on unload (your problem). 

I'm going to guess #2 because from the end of the fast bowden move your toolhead sensor is found after moving 9.3mm.  I.e. the filament WAS tight against the extruder gears.

Try backtracking:
Recalibrate GEAR,
Then recalibrate ENCODER
Then recalibrate BOWDEN_LENGTH.

BTW What are you running on?  560mm is quite a short bowden (mine is 680 on a Voron 2.4 with shortest path I could get).

protorayn, 2024-07-30:
I actually did that calibration before I posted here. I will do again and reset the parking distance to default. 

As far as the extruder sensor distance, I had done the automatic toolhead calibration now that 2.6 is out. So not sure why that was so tight. The measured distance is 16.3mm. 

I run a steathburner on an OG Ender 3. This machine is my tinker printer. Little more than just the frame and motors left at this point. The ERCF is mounted on top (though I might move it to a shelf above, the cotton tail makes it pretty heavy).
Print height is only 250 so 560mm seems long to me, but that is what keeps it from bending too sharply.

protorayn, 2024-07-30:
So I rebuilt the encoder once more to make sure it had a good encoder disk. (The original disk didn't look so good.) There was some slippage on the mmu gear so I tightened that a bit.

Re-ran all the calibration and reset the default park pos. Seems to have done the trick. I ran a loading sequence soak test and all seemed fine. Runnning a test print now to see how far it gets. 

Thanks for taking the time to look. 

I'll close this thread as it seems to be resolved.

closed 2024-07-30 by protorayn