moggieuk/Happy-Hare #358: Feature Request: variable to set stepper current for cutting motion

protorayn, 2024-07-28:
I am using a Filametrix on a standard bedslinger (OG Ender 3) but the X axis stepper skips at normal printing current. 

I have edited the MMU_CUT_TIP macro to adjust my stepper current before and after the _DO_CUT_MOTION calls. 

This of course throws an error in mainsail because this file is tracked in git and is now dirty. So anytime I update I have to re-edit the file in order for it to work. 

A variable to set the current for either the X or Y for printers that are non-standard voron/core-xy type machines would solve this issue.

moggieuk, 2024-07-30:
The Mainsail "error" isn't an error -- it is just reminding you that you have deviated from the default.

You can edit but just know you will have to add back after an update.

I would be interested in a clean PR for this enhancement...  if should probably be done through a "pre_cut_user_extension" because which movement stepper is specific to mounting.

Durahl, 2024-08-10:
Not sure if this would even be a solution to your problem since I've never dabbled with the Filametrix ( It's just too ugly ) but maybe try if you have more luck with a #11 Blade that should require less force due to its angled profile? 

It's what I'm using for my Tool Head Cutter...
[image]
...though arguably I'm also using beefier Stepper Motors 😅

ningpj, 2026-07-05:
Old issue. Stepper current changes supported for defined x/y cutting axis