moggieuk/Happy-Hare #360: Delay Variable for Pre-Gate Sensor Preload Function

SkiBikePrint, 2024-07-29:
For pre-gate sensors that are remote from the MMU, such as at the buffer, it would be useful to have a user set delay variable for the automatic preload function.  This gives the user time to push filament to the mmu as part of loading a new spool.  Default could be zero which would maintain the current functionality of an immediate attempt to preload.

moggieuk, 2024-07-30:
Rather than delay, there is a new feature in v2.7.0  `preload_attempts` (default 5).  This can be increased and consequently the time will increase.

SkiBikePrint, 2024-07-30:
Even better!  Thanks!

moggieuk, 2026-10-21:
Closing this as it was sorted out in the thread - the preload retries do the job of a delay.

In Happy Hare v4 the option is now called `gate_preload_attempts` in `mmu_parameters.cfg` (menuconfig: "Filament preloading and ejection" > "Number of preload attempts"). Raise it to give yourself more time to push the filament through from a remote buffer. Automatic loading when the entry sensor triggers is controlled by `gate_autoload` in the same menu.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk