moggieuk/Happy-Hare #361: Be able to run HH w/ LEDs only using native Klipper neopixel support

Sparkss, 2024-07-30:
Enhancement Request: To remove the need to have the klipper-led_effect module installed to run HH w/ LEDs. It would be nice to remove the led_effect requirement check, add an optional variable for led_effects or native neopixel and update the macros in mmu_leds.cfg to allow the option to just use native Klipper neopixel support/commands. While it would not provide the same level of visual functionality as the led_effect module does, it would reduce the load on the MCUs.

Rationale: The TTC issues appears to be exacerbated by processor intensive routines. Everything I have read around performance metrics show the led_effect module as 1. being rather resource intensive and 2. anecdotal evidence points to a memory or other resource leak in the module that builds up over time until it reaches a tipping point/threshold.

moggieuk, 2024-07-30:
This is a good idea.  Would love a PR for this from somebody.  It's on my wishlist, but not at the top at the moment and I'm swamped with work and requests.

justinr54321, 2024-08-02:
I have found two things that help this:
Reinstall the LED effects after you install HH and to raise the frame rate to 30. For some reason helps to take up less resources, while a lower frame rate takes up more.
Currently in a middle of a print, and all is going well.

[image]

moggieuk, 2024-08-07:
Frame rate decreases load... that is very strange.  I need to look into that.

BTW. Static LED option is not available in v2.7.0beta (soon to be main).  This makes the led effects module an option but also allows turning off animation even if it is installed.

```yml
variable_led_animation_enable   : True                  ; Whether to use led-animation-effects vs static LEDs
```
This can turn off animation of LEDS (to give more head room to those suffering from TTC).  Also allows for LEDS without the klipper_led_effects being installed.

So marking this issue as resolved.

Sparkss, 2024-08-14:
Sawadee-krub, Thank you very much. I was going to work on the LED script but have been just buried at work. I appreciate your working it into the next release, as I am sure others do as well.

closed 2024-08-14 by Sparkss