moggieuk/Happy-Hare #364: "Internal error during connect: name 'allVariables' is not defined" - After recent upgrade

JoaquinBerrios, 2024-07-30:
I updated Klipper and Happy Hare in the last couple of days and I'm getting the above error. Here is as snippet from my klippy.log file with additional details. I can provide other logs as well. 

```
mcu 'mmu': got {'oid': 2, 'next_clock': 4112041947, 'count': 0, 'count_clock': 0, '#name': 'counter_state', '#sent_time': 322.572227956, '#receive_time': 322.590026585}
mcu 'EBBCan': got {'oid': 11, 'next_clock': 4130711314, 'value': 7577, '#name': 'analog_in_state', '#sent_time': 322.51266621499997, '#receive_time': 322.612889974}
mcu 'EBBCan': got {'oid': 12, 'next_clock': 4131351314, 'value': 30882, '#name': 'analog_in_state', '#sent_time': 322.51266621499997, '#receive_time': 322.62289578900004}
mcu 'mcu': got {'oid': 18, 'next_clock': 613759744, 'value': 31185, '#name': 'analog_in_state', '#sent_time': 322.513357104, '#receive_time': 322.679155937}
mcu 'mmu': got {'oid': 2, 'next_clock': 4118505947, 'count': 0, 'count_clock': 0, '#name': 'counter_state', '#sent_time': 322.572227956, '#receive_time': 322.691018919}
mcu 'mcu': got {'oid': 20, 'next_clock': 617359744, 'value': 7662, '#name': 'analog_in_state', '#sent_time': 322.513357104, '#receive_time': 322.69915577}
mcu 'mmu': got {'oid': 2, 'next_clock': 4124969947, 'count': 0, 'count_clock': 0, '#name': 'counter_state', '#sent_time': 322.572227956, '#receive_time': 322.792024085}
mcu 'mmu': got {'oid': 2, 'next_clock': 4131433947, 'count': 0, 'count_clock': 0, '#name': 'counter_state', '#sent_time': 322.572227956, '#receive_time': 322.893023863}
mcu 'EBBCan': got {'oid': 11, 'next_clock': 4149911314, 'value': 7580, '#name': 'analog_in_state', '#sent_time': 322.51266621499997, '#receive_time': 322.912887011}
mcu 'EBBCan': got {'oid': 12, 'next_clock': 4150551314, 'value': 30892, '#name': 'analog_in_state', '#sent_time': 322.51266621499997, '#receive_time': 322.922890789}
mcu 'mcu': got {'oid': 18, 'next_clock': 667759744, 'value': 31186, '#name': 'analog_in_state', '#sent_time': 322.513357104, '#receive_time': 322.979153178}
mcu 'mmu': got {'oid': 2, 'next_clock': 4137897947, 'count': 0, 'count_clock': 0, '#name': 'counter_state', '#sent_time': 322.572227956, '#receive_time': 322.994015215}
Loaded MCU 'scanner' 73 commands (CARTOGRAPHER 4.1.0 - BETA -  / 29-04-2024 17-38)
MCU 'scanner' config: ADC_MAX=4095 BUS_PINS_i2c1=PB6,PB7 BUS_PINS_spi1=PA6,PA7,PA5 CANBUS_FREQUENCY=1000000 CARTOGRAPHER_ADC_SMOOTH_COUNT=16 CLOCK_FREQ=48000000 MCU=stm32f042x6 PWM_MAX=2 RECEIVE_WINDOW=192 RESERVE_PINS_CAN=PA11,PA12 RESERVE_PINS_crystal=PF0,PF1 STATS_SUMSQ_BASE=256 STEPPER_BOTH_EDGE=1
Configured MCU 'mcu' (1024 moves)
Configured MCU 'mmu' (1024 moves)
Configured MCU 'EBBCan' (1024 moves)
Configured MCU 'rpi' (1024 moves)
  DEBUG: Found tmc2209 on selector_stepper. Stallguard 'touch' homing possible.
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 135, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/mmu.py", line 809, in handle_connect
    self.gear_tmc = self.printer.lookup_object('%s %s' % (chip, self.GEAR_STEPPER_CONFIG), None)
                    ^^^^
NameError: name 'allVariables' is not defined
Resetting prediction variance 1395.655: freq=49996325 diff=120264 stddev=5098.848
Resetting prediction variance 2418.572: freq=49997272 diff=173837 stddev=3224.705
```

moggieuk, 2024-07-31:
That is a strange error because the reported line number cannot generate that error!   I have to suspect an upgrade problem (like the `[save_variables]` file could not be found or similar).   If you have a more complete `klippy.log` I can take a look but if this were a problem everyone would experience it but you are the only one that have reported...

JoaquinBerrios, 2024-07-31:
It's standard Klipper although I am participating in the Cartographer tap beta, and have some "untracked sources", most of which are from HH, but a couple that are not (gcode shell command led effects). That said the restored HH backup worked fine without having to change anything else. 

Let me try pulling the latest versions of everything and running through the update script again in case it was just an issue with that specific upgrade run. If that still errors out and I can't chase the issue down I can provide a complete log file with any additional details I find while troubleshooting.

JoaquinBerrios, 2024-07-31:
OK, so reverted to a previous commit, re-updated and everything seems to be working fine now so I think it was just a bad update. Sorry to have bothered you. Maybe should run fsck on that SSD just to be sure...