moggieuk/Happy-Hare #366: Position Depressor pin at specific Z hight when using Filametrix

iwm911, 2024-08-01:
Hi,

i have a Switchwire with Filametrix cutter.

i want to mount the depressor pin at max Z instead of the X beam.
the ideas is to not lose any space on the X axis (it's small as it is) and instead lose some space on the Z axis (where it is rarely used to the max)

in a scenario where the mount will be fixed at the max Z height  the gantry will move to a safe location (i.e no at the depressor X), move to Z height specified, and then move to the depressor X, cut the filament and return to print.

on the switchwire Z moves can be pretty fast so it shouldn't take too much time.

any way to implement it in the current firmware or will it require some development?

looked at the docs with no luck.

thanks!

moggieuk, 2024-08-07:
The "cut_tip" macro could be enhanced with more options (would like a PR for that because it isn't my focus at the moment).  That said, the best way would be for you to implement the tip cutting macro using the current one as a base:

copy _MMU_CUT_TIP macro into your own "cfg" file, Change macro names so it doesn't collide with default one. THEN change this in `mmu_parameters.cfg` to point to your version of the tip cutting macro:
```yml
form_tip_macro: _YOUR_FORM_TIP            # Name of macro to call to perform the tip forming (or cutting) operation
```
This will survive upgrades and you have your own custom version of part of the tool-change process.

Note, you can also inject single macro calls using:
```
variable_user_pre_unload_extension  : ''        ; Executed after default logic
variable_user_post_unload_extension : ''        ; Executed after default logic
variable_user_pre_load_extension    : ''        ; Executed after default logic
variable_user_post_load_extension   : ''        ; Executed after default logic but before restoring toolhead position
variable_user_park_move_macro       : ''        ; Executed instead of default 'G1 X Y move' to 'variable_park_xy' position
```
But I don't think you will get the level of functionality you want with these macro hooks. Better to point to entirely new tip cutting macro...

moggieuk, 2024-08-08:
To be clear. If you update the current macro a PR would be helpful so I can easily integrate.  If a completely custom macro then you wouldn't want to create PR..

closed 2024-11-13 by github-actions[bot]