moggieuk/Happy-Hare #368: Internal Error on Command: "SET_LED_EFFECT"

justinr54321, 2024-08-02:
Setting up a chain of LEDs from the board to the ERCT LEDs, then to the Apron LEDs, and I get the error mentioned above. Klippy log shows it to be an issue with the python files. Showing no errors in the MMU log.
[Log Files.zip](https://github.com/user-attachments/files/16473201/Log.Files.zip)
[image]
When I comment out the exit range, no issues except for no Apron LEDs.

moggieuk, 2024-08-07:
I'd recommend a lower frame rate: the default 24 is already very high.
What does this section look like:
```
[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 18                        # Number gates x1 or x2 + 1 (if you want status)
color_order: GRBW  
```

justinr54321, 2024-08-07:
> What does this section look like:
> 
> ```
> [neopixel mmu_leds]
> pin: mmu:MMU_NEOPIXEL
> chain_count: 16                        # Number gates x1 or x2 + 1 (if you want status)
> color_order: GRBW  
> ```
> 
> I wonder if you LED chain isn't long enough.. would need to be at least 16 LEDs.
> 
> Also, make sure your klipper-led_effects module is up-to-date because the failure is coming from within that module.

Thank you. I forgot to change it from 8 to 16. And is now working. 🤦‍♂️

closed 2024-08-07 by justinr54321

moggieuk, 2024-08-07:
Cool!