moggieuk/Happy-Hare #371: How to configure multiple drive motors?

3DCoded, 2024-08-09:
I recently created the [3MS](https://github.com/3DCoded/3MS), a modular multimaterial system for Klipper. Currently it has custom macros developed by me, but I want to control it via HappyHare for more robust functionality. I am a bit lost about how to configure multiple drive motors (each motor has its own driver) and no selector/servo. Any help would be appreciated!

moggieuk, 2024-08-09:
That functionality is being worked on as we speak.  A lot of work is coded in the current release but not exposed. Stay tuned...

3DCoded, 2024-08-09:
Thank you for the quick reply. I'll stay tuned. If you need any help testing, let me know.

moggieuk, 2024-08-11:
Thanks. I have a lot of folks waiting for multi-drive motor support.  I have my test system operational as of yesterday so can now get back to writing the missing code.   Really configuration is going to be the biggest challenge...  (to keep backward compatibility for type-A designs).  I'm thinking about a new approach to `mmu_hardware.cfg`

Podster2020, 2024-10-09:
hi is the type b had any love recently? im mid way through a build and would like to use type b

moggieuk, 2024-12-15:
Happy Hare version 3.0 introduces this functionality so closing issue.

closed 2024-12-15 by moggieuk