moggieuk/Happy-Hare #372: Option 'spoolman_support' is not valid in section 'mmu'

phnxdrgn536, 2024-08-09:
i am trying to follow the "configurations" section of the wiki and i get two errors. Option 'spoolman_support' is not valid in section 'mmu' pops up whenever i boot and it prevents boot from continuing and i get the red klipper error in mainsail and tells me to fix the problem and restart. the second error i get is a moonraker warning saying "unparsed config option 'update_spoolman_location: true' detected in [mmu_server] section.

current version 
SPOOLMAN: v0.19.0 (c93ca5d)
MOONRAKER: v0.9.2-4-g30ac5dfa
HAPPY HARE: v2.6.0-60-g2e2ce32a
KLIPPER: v0.12.0-276-gca815f52

moggieuk, 2024-08-09:
The doc is slightly ahead (can't have different doc branches).  The v270beta branch operates exactly as the doc, the current main v2.6.0 branch uses `enable_spoolman: 0 | 1`

You can opt into the v2.7.0beta with:
```
cd ~/Happy-Hare
./install.sh -b v270beta
```

Just remember when I make that live (within a week) you will need to run:
```
cd ~/Happy-Hare
./install.sh -b main
```
to switch back.

phnxdrgn536, 2024-08-09:
oh ok, thanks for the explanation. if its gonna be live in a week or so I'll just wait for that

closed 2024-08-09 by phnxdrgn536