moggieuk/Happy-Hare #373: Changing the cutting axis to Y

EdwinPock, 2024-08-09:
If the cutting movement in my printer is in the Y axis
Just change into mmm_cut_tip.cfg?

G1 X{fast_slow_transition_loc} F{cut_fast_move_speed * 60} # Fast move to initiate contact of the blade with filament
 G1 X{pin_loc_x_compressed} F{cut_slow_move_speed * 60} # Do the cut in slow move
to
 G1 Y{fast_slow_.....
 G1 Y{pin_loc_x_compres......
?

moggieuk, 2024-08-10:
The cutting macro could do with some more options.  If you edit it directly it will be overwritten by upgrade (it is designed to be read only).  You can duplicate it, rename it and call you custom cut macro rather than the default one. There is some info on that here: https://github.com/moggieuk/Happy-Hare/wiki/Macro-Customization

I'd definitely be interested in a PR for enhance the cutting macro..

EdwinPock, 2024-08-10:
Will something like this work?

Changes in [gcode_macro _mmu_cut_tip_vars]:
# This should be the X or Y axis a cutting mechanism
variable_cutting_axis				: x 		; x or y axis

# This distance is added to 'pin_loc_x' to determine the starting position
# and to create a small saftely distance that aids in generating momentum

#######variable_pin_park_x_dist        : 5.0		; Distance in mm
variable_pin_park_dist        : 5.0		; Distance in mm

# Position of the toolhead when the cutter is fully compressed. Should leave# a small headroom (should be a bit larger than 0, or whatever xmin is) to
# avoid banging the toolhead or gantry

#######variable_pin_loc_x_compressed   : 5		; x coordinate
variable_pin_loc_compressed   : 5		; x or y coordinate

Changes in [gcode_macro _FILAMETRIX_DO_CUT_MOTION]:

[gcode_macro _FILAMETRIX_DO_CUT_MOTION]
description: Helper to do a single horizontal cut movement
gcode:
    {% set pin_park_x_loc = params.PIN_PARK_X_LOC | float %}
	{% set pin_park_y_loc = params.PIN_PARK_Y_LOC | float %}
    {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
	
	{% set cutting_axis = vars['cutting_axis']|string %}
	{% set pin_loc_compressed = vars['pin_loc_compressed']|float %}
	{% set cut_fast_move_fraction = vars['cut_fast_move_fraction']|float %}
    {% set cut_fast_move_speed = vars['cut_fast_move_speed']|float %}
    {% set cut_slow_move_speed = vars['cut_slow_move_speed']|float %}
    {% set cut_dwell_time = vars['cut_dwell_time']|float %}
    {% set evacuate_speed = vars['evacuate_speed']|float %}
    {% set rip_length = vars['rip_length']|float %}
    {% set rip_speed = vars['rip_speed']|float %}
	
	# {% if cuting_axis == "x" %}
	# 	{% set fast_slow_transition_loc = (pin_loc_compressed - pin_park_x_loc) * cut_fast_move_fraction + pin_park_x_loc|float %}
	# {% elif cuting_axis == "y" %}
	#	{% set fast_slow_transition_loc = (pin_loc_compressed - pin_park_y_loc) * cut_fast_move_fraction + pin_park_y_loc|float %}
	# {% endif %}
	# or
	
	set fast_slow_transition_loc = (pin_loc_compressed - pin_park_{cutting_axis}_loc) * cut_fast_move_fraction + pin_park_{cutting_axis}_loc|float %}
	

    G1 {cutting_axis}{fast_slow_transition_loc} F{cut_fast_move_speed * 60} # Fast move to initiate contact of the blade with filament
    G1 {cutting_axis}{pin_loc_compressed} F{cut_slow_move_speed * 60} # Do the cut in slow move
    G4 P{cut_dwell_time}
    {% if rip_length > 0 %}
        G1 E-{rip_length} F{rip_speed * 60}
    {% endif %}
	
    G1 {cutting_axis}{pin_park_{cutting_axis}_loc} F{evacuate_speed * 60} # Evacuate

###########################################################################

EdwinPock, 2024-08-10:
@moggieuk  Can something like this work?
Changes in
[gcode_macro _MMU_CUT_TIP_VARS]
	
		
	# This should be the X or Y axis a cutting mechanism
	variable_cutting_axis				: x 		; x or y axis
	
	# This distance is added to 'pin_loc_x' to determine the starting position
	# and to create a small saftely distance that aids in generating momentum
	
	#######variable_pin_park_x_dist        : 5.0		; Distance in mm
	variable_pin_park_dist        : 5.0		; Distance in mm
	
	# Position of the toolhead when the cutter is fully compressed. Should leave# a small headroom (should be a bit larger than 0, or whatever xmin is) to
	# avoid banging the toolhead or gantry
	
	#######variable_pin_loc_x_compressed   : 5		; x coordinate
	variable_pin_loc_compressed   : 5		; x or y coordinate

Changes in:
[gcode_macro _FILAMETRIX_DO_CUT_MOTION]

	[gcode_macro _FILAMETRIX_DO_CUT_MOTION]
		[gcode_macro _FILAMETRIX_DO_CUT_MOTION]
	description: Helper to do a single horizontal cut movement
	gcode:
    		{% set pin_park_x_loc = params.PIN_PARK_X_LOC | float %}
    		{% set pin_park_y_loc = params.PIN_PARK_Y_LOC | float %}
    		{% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
	# new string	
    		{% set cutting_axis = vars['cutting_axis']|string %}
    		{% set pin_loc_compressed = vars['pin_loc_compressed']|float %}
    		{% set cut_fast_move_fraction = vars['cut_fast_move_fraction']|float %}
    		{% set cut_fast_move_speed = vars['cut_fast_move_speed']|float %}
    		{% set cut_slow_move_speed = vars['cut_slow_move_speed']|float %}
    		{% set cut_dwell_time = vars['cut_dwell_time']|float %}
    		{% set evacuate_speed = vars['evacuate_speed']|float %}
    		{% set rip_length = vars['rip_length']|float %}
    		{% set rip_speed = vars['rip_speed']|float %}
	
	# {% if cuting_axis == "x" %}
	#     {% set fast_slow_transition_loc = (pin_loc_compressed - pin_park_x_loc) * cut_fast_move_fraction + pin_park_x_loc|float %}
	# {% elif cuting_axis == "y" %}
	#     {% set fast_slow_transition_loc = (pin_loc_compressed - pin_park_y_loc) * cut_fast_move_fraction + pin_park_y_loc|float %}
	# {% endif %}
	
	# or 
	#I don't know if you can :-)
	
	set fast_slow_transition_loc = (pin_loc_compressed - pin_park_{cutting_axis}_loc) * cut_fast_move_fraction + pin_park_{cutting_axis}_loc|float %}  ## 
	
    G1 {cutting_axis}{fast_slow_transition_loc} F{cut_fast_move_speed * 60} # Fast move to initiate contact of the blade with filament
    G1 {cutting_axis}{pin_loc_compressed} F{cut_slow_move_speed * 60} # Do the cut in slow move
    G4 P{cut_dwell_time}
    {% if rip_length > 0 %}
        G1 E-{rip_length} F{rip_speed * 60}
    {% endif %}
	
    G1 {cutting_axis}{pin_park_{cutting_axis}_loc} F{evacuate_speed * 60} # Evacuate

moggieuk, 2024-08-11:
It looks like it is going in the right direction.  May not want to duplicate all variable with "x" or "y", but rather just interpret a non direction-specific distance based on your "direction" setting.

EdwinPock, 2024-08-23:
I finally changed in
mmm_macro_vars.cfg:
```g-code
# This should be the position of the toolhead where the cutter arm just
# lightly touches the depressor pin
variable_pin_loc_xy             : 50, 204	; x,y coordinates of depressor pin

# In which axis does the extruder move to make a cut
variable_pin_loc_axis			: "y"			; x or y axis

# Orientation on the axis where the cutting mechanism is mounted
# If the cut is performed at the minimum value of the axis, set it to min
# If cutting is performed at the maximum axis value, set it to max
variable_pin_axis_orientation	: "max"		; min = Min position axis, max = Max position axis

# This distance is added to 'pin_loc_x' to determine the starting position
# and to create a small saftely distance that aids in generating momentum
#variable_pin_park_x_dist        : 5.0		; Distance in mm
variable_pin_park_dist        : 5.0		; Distance in mm

# Position of the toolhead when the cutter is fully compressed. Should leave
# a small headroom (should be a bit larger than 0, or whatever xmin is) to
# avoid banging the toolhead or gantry
#variable_pin_loc_x_compressed   : 0.5		; x coordinate
variable_pin_loc_xy_compressed   : 50, 209  ; x,y coordinates of fully compressed
```g-code
``` 

Checked works, of course after each update you have to re-introduce macro_vars :-)
Maybe it will be useful to someone. You need to change the name of your own macro.
I added my macro tip_cut2.cfg :
```g-code
########################################################################################################################
# Happy Hare supporting macros
#   Standalone Tip Cutting for "Filametrix" style toolhead cutters
#
# THIS FILE IS READ ONLY
#
# To configure, set
#   'form_tip_macro: _MMU_CUT_TIP' in 'mmu_parameters.cfg'
#
# IMPORTANT:
#   The park position of the filament is relative to the nozzle and
#   represents where the end of the filament is after cutting. The park position
#   is important and used by Happy Hare both to finish unloading the extruder
#   as well as to calculate how far to advance the filament on the subsequent load.
#
# When using this macro it is important to turn off tip forming in your slicer and
# force Happy Hare to always run this when loading filament by adding:
#   'force_form_tip_standalone: 1' in 'mmu_parameters.cfg'
# Also decide on 'return_to_wipetower' vs 'enable_park' (mmu_macro_vars.cfg)
#
# Implementation note: It is important to report back the position the cutter
# leaves the filament in the extruder via the variable 'output_park_pos' so this
# is set dynamically in gcode with this construct:
#   SET_GCODE_VARIABLE MACRO=_MMU_CUT_TIP VARIABLE=output_park_pos VALUE=..
#
[gcode_macro _MMU_CUT_TIP2]
description: Cut filament by pressing the cutter on a pin with a horizontal movement

# -------------------------- Internal Don't Touch -------------------------
variable_output_park_pos: 0 # Dynamically set in this macro

gcode:
    {% set final_eject = params.FINAL_EJECT|default(0)|int %}
    {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
    {% set pin_loc_x, pin_loc_y = vars.pin_loc_xy|map('float') %}
    {% set pin_park_dist = vars['pin_park_dist']|float %}
    {% set retract_length = vars['retract_length']|float %}
    {% set simple_tip_forming = vars['simple_tip_forming']|default(true)|lower == 'true' %}
[... 166 lines cut ...]
description: Operate optional gantry servo operated pin
gcode:
    {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
    {% set gantry_servo_enabled = vars['gantry_servo_enabled']|default(true)|lower == 'true' %}
    {% set angle = vars['gantry_servo_down_angle']|float %}

    {% if gantry_servo_enabled %}
        SET_SERVO SERVO=mmu_gantry_servo ANGLE={angle}
    {% endif %}

###########################################################################
# Helper macro for tip cutting
#
[gcode_macro _FILAMETRIX_GANTRY_SERVO_UP]
description: Operate optional gantry servo operated pin
gcode:
    {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
    {% set gantry_servo_enabled = vars['gantry_servo_enabled']|default(true)|lower == 'true' %}
    {% set angle = vars['gantry_servo_up_angle']|float %}

    {% if gantry_servo_enabled %}
        SET_SERVO SERVO=mmu_gantry_servo ANGLE={angle}
    {% endif %}

```

moggieuk, 2024-09-04:
Read this comment: https://github.com/moggieuk/Happy-Hare/pull/413#issuecomment-2328011267

This is available now in the beta v2.7.1 branch (aka "auto-calibrate"