moggieuk/Happy-Hare #374: Question , cant find any info or answers

feaver12, 2024-08-11:
Sorry to ask like this but I have a question that I'm  struggling to find any info on....and nobody i ask can seem to give me any awnser to - under parameter "**gate_endstop_to_encoder**" what do you mean by "**+ve if encoder after endstop**"? 
I keep getting failed to unload errors when retry unload is called, but the filament laterally falls just short of the gate endstop by a few mm (maybe 5mm max). I have messed with parameter values relating to gate-encoder distance a little up to 20mm difference in values and there is no change. I'm guessing my issue has something to do with this as my "gate endstop" is at the end of a splitter and the encoder is fitted after the gate endstop.... thank you in advance.

moggieuk, 2024-08-11:
This setting is purely to describe the distance from the gate endstop (sensor after the gate) to the encoder. It is only used if encoder is fitted and originated when Tradrack owners fitted an encoder after the existing gate sensor.  In that situation it is a positive value.  In a MMU design where the optional encoder was before the gate sensor it would be negative.

Note that when unloading filament Happy Hare will "home" to whatever the defined gate endstop is.  This can be a switch (gate sensor) or use the encoder.  This is set with `gate_homing_endstop`.

`MMU_STATUS SHOWCONFIG=1` is really helpful to explain the load/unload sequence used based on your current configuration.

feaver12, 2024-08-11:
Ok that makes sense. I am using a custom mmu that is not much like the
ercf. So what value should I be looking at? The filament unloads fine until
there is a load error, when the loading error occurs, upon unloading (ive
noticed at a much slower speed) the filament falls short of the gate
endstop by about 5mm....I have messed with gate homing max and gate endstop
to encoder values (see attachment) and nothing seems to work. The filament
always falls short of the gate/splitter endstop by about 5mm but only when
an error is thrown. Never under normal unloading conditions....

moggieuk, 2024-08-15:
The slow speed is deliberate because the previous error means HH isn't sure where the filament end is and so it moving slower.  If using an encoder to unload this results in lots of small movements.  If using a gate sensor then it is a continuous slow homing speed.

If this stops short of the gate, then either the calibrated bowden length is too short or your `gate_homing_max` is too short.

feaver12, 2024-08-15:
Ok. Thank you for taking the time to respond. So I'm guessing that if
everything is loading correctly 99% of the time then that means the bowden
length is pretty close so I should be increasing gate_homing_max? I thought
i had tried that but maybe i didn't increase it enough. Would any harm be
caused if i increase the value too much? I calibrated bowden length to gate
sensor using the manual method (pushed filament all the way to extruder
gears manually and run the command).

feaver12, 2024-08-19:
I have increased gate_homing_max  as well as calibrated_bowden_distance as
mentioned earlier and have done so in increments of 5 by upto nearly 100mm
and still the filament falls just short of the gate sensor by about 5mm in
the exact same spot  Nothing changes. It only happens when there is a
failed to load error and the mmu retries loading - this is when it fails to
unload. Any ideas? Everything else works perfectly.  It's only when I have
a failed load that this occurs

moggieuk, 2024-09-17:
Ok. If it is not parking at the correct distance then you can adjust with `gate_parking_distance`.  Are you using a ERCF or other MMU?

feaver12, 2024-09-17:
I am using a custom mmu. I managed to rectify it by adding 2mm to my
calibrated bowden length...
Also I have replied to you on discord (feva) with a description of the
issues I am having after updating

closed 2024-11-13 by github-actions[bot]