moggieuk/Happy-Hare #376: Add configurable delay between tip cutting servo enabling and cutting motion

HitLuca, 2024-08-12:
After installing a servo cutter on my trident gantry, I noticed that the toolhead did the cutting motion too early, when the servo wasn't engaged fully. After checking the code I believe there's no way to delay the cutting motion, so I had to modify the tip cutting macro myself. Adding a configurable delay would handle slower/faster servos, and always result in the cutting arm being properly engaged.

My change was adding a `G4` command in the `_MMU_CUT_TIP` macro, between these lines

```
[gcode_macro _MMU_CUT_TIP]
[...]
    _FILAMETRIX_GANTRY_SERVO_UP
    _FILAMETRIX_MOVE_TO_CUTTER_PIN PIN_PARK_X_LOC={pin_park_x_loc} PIN_PARK_Y_LOC={pin_park_y_loc}
    _FILAMETRIX_GANTRY_SERVO_DOWN
    G4 P500 <- 500ms wait
    _FILAMETRIX_DO_CUT_MOTION PIN_PARK_X_LOC={pin_park_x_loc} RIP_LENGTH={rip_length}
    _FILAMETRIX_GANTRY_SERVO_UP
[...]
```

Having the `P500` parameter being configurable would be better in my opinion, to not slow down faster ones

I'm pretty sure I can get all the functionality implemented myself, let me know if you want a pull request made

moggieuk, 2024-08-15:
That is a really good point.  I can add this.  Honestly most folks don't have servo, they just have the pin.

HitLuca, 2024-08-15:
Great!

HitLuca, 2024-08-20:
Quick update: I noticed that my gantry servo was defective and opened much more slowly than a new one, so technically I don't need the delay anymore. I would still consider this a good improvement to account for weird setups

moggieuk, 2024-09-04:
Added as precaution in "auto-calibrate" branch (aka 2.7.1 beta)

moggieuk, 2026-10-21:
Closing this as it was added back in v2.7.1 and is still in Happy Hare v4. When the gantry servo is enabled (`variable_gantry_servo_enabled` in `mmu_macro_vars.cfg`), the cut macro now waits a fixed 500ms after deploying the servo pin before it starts the cutting move, so a normal servo has time to fully engage. The wait isn't configurable. From your last comment it sounds like the replacement servo fixed it for you anyway.

If there's something v4 still doesn't do for you, please open a new feature request against v4 - it's being actively worked on.

closed 2026-10-21 by moggieuk