moggieuk/Happy-Hare #380: EREC cutting befor Ejecting filament

mochoandre, 2024-08-14:
Hello, I'm trying to tune my EREC filament Cutter but it's going though.

I have 
```
variable_user_pre_unload_extension  : 'BLOBIFIER_PARK'	; Executed after default logic
variable_user_post_unload_extension : 'EREC_CUTTER_ACTION'	; Executed after default logic
variable_user_pre_load_extension    : 'BLOBIFIER_PARK'	; Executed after default logic
variable_user_post_load_extension   : 'BLOBIFIER'	; Executed after default logic but before restoring toolhead position
```

But HH is cutting filament when still fully loaded in bowden and of course fillament gets stucked inside bowden.
How can I change this behaviour?

moggieuk, 2024-08-15:
You would need to supply your `mmu.log`, because that setup looks correct and 
"variable_user_post_unload_extension" macro is only called AFTER the unload.

checkitog, 2024-08-16:
Friend, I also have Erec and in my case it makes perfect cuts when the filament has already returned from the hotend, check your configuration

mochoandre, 2024-08-26:
Sorry for the late reply but I've been away of home .
Updated to 2.70 and now it works good.

Thank you

mochoandre, 2024-08-28:
I thought My issue was solved with upgrade to 2.70 but it's not. 
I've done a print and after some hours I issued another tool to start a new print from an old file (without mmu slicer settings).
As soon as I issued tool change EREC cut filamment with it still loaded in tube.

Here is the log

```
03:18:03 > _MMU_PRINT_END STATE=complete
03:48:18 > _MMU_PRINT_END STATE=standby
03:48:18   DEBUG: Job State: COMPLETE -> STANDBY (MMU State: Encoder: Disabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 1800.00s)
17:58:14 > MMU_CHANGE_TOOL TOOL=3
17:58:14   DEBUG: Job State: STANDBY -> IDLE (MMU State: Encoder: Disabled, Synced: False, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 1800.00s)
17:58:14   DEBUG: Saving toolhead gcode state and position (X:175.0 Y:175.0 Z:3.0 E:-2.0) for change_tool
17:58:14   DEBUG: Tool change initiated with standalone MMU tip forming
17:58:14 Tool change requested, from T0 to T3
17:58:14   DEBUG: Unloading tool T0
17:58:21 Unloading filament...
17:58:21 [T0] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
17:58:21   DEBUG: Setting servo to down (filament drive) position at angle: 30
17:58:21 > MMU_STATS COUNTER=servo_down INCR=1
17:58:21   DEBUG: Syncing gear stepper and extruder
17:58:22 Forming tip...
17:58:24 > _MMU_STEP_MOVE MOVE=58
17:58:24   DEBUG: Moving 'gear' motor 58.0mm...
17:58:24   DEBUG: Unsyncing gear stepper and extruder
17:58:30 > _MMU_STEP_MOVE MOVE=-1
17:58:30   DEBUG: Moving 'gear' motor -1.0mm...
17:58:32 > MMU_STATS COUNTER=cutter_blade INCR=1
17:58:32 > _MMU_STEP_SET_FILAMENT STATE=2
17:58:32 [T0] < En <........ [Ex .. (*) .. Nz] 57.0mm (e:61.2mm)
17:58:32 > _MMU_STEP_UNLOAD_GATE
17:58:32   DEBUG: Slow unload of the encoder
17:58:34 [T0] < En ......... [Ex .. (*) .. Nz] UNLOADED 10.1mm (e:95.1mm)
17:58:34 > _MMU_M400
17:58:34 [T0] < En <<<<<<<<< [Ex << (*) <. Nz] -16.0mm (e:18.4mm)
17:58:34   DEBUG: Extracting filament from extruder
17:58:34   DEBUG: Reverse homing up to 41.4mm to toolhead sensor (synced)
17:58:37 MMU issue detected whilst out of a print
         Reason: Unload sequence failed: Failed to reach toolhead sensor after moving 41.4mm.
         Occured when changing tool: T0 > T3
17:58:37   DEBUG: Setting servo to up (filament released) position at angle: 140

```

Could It be that my motor Angles are inverted ? to proper align it I have to use like  

```
variable_servo_closed_angle   : 0	; Servo angle for closed postion with bowden aligned MMU
variable_servo_open_angle     : 180	; Servo angle to open up the cutter and move bowden away from MMU
```

insted of the example like

```
variable_servo_closed_angle   : 70    ; Servo angle for closed postion with bowden aligned MMU
variable_servo_open_angle     : 0     ; Servo angle to open up the cutter and move bowden away from MMU
```

Best Regards

mochoandre, 2024-08-28:
Better Log 

```
18:51:25 > MMU_CHANGE_TOOL TOOL=3
18:51:25   DEBUG: Cannot save toolhead position or z-hop for change_tool because not homed
18:51:25   DEBUG: Tool change initiated with standalone MMU tip forming
18:51:25     TRACE: Processing idle_timeout 'printing' event
18:51:25 Tool change requested, from T0 to T3
18:51:25   DEBUG: Unloading tool T0
18:51:25     TRACE: Running macro: _MMU_PRE_UNLOAD
18:51:25 Unloading filament...
18:51:25     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Idle'
18:51:25     TRACE: (toolhead sensor detects filament)
18:51:25 [T0] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:0.0mm)
18:51:25     TRACE: (toolhead sensor detects filament)
18:51:25     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Forming Tip' OLD_ACTION='Unloading'
18:51:25   DEBUG: Setting servo to down (filament drive) position at angle: 30
18:51:25     TRACE: Running macro: _MMU_EVENT EVENT=servo_down 
18:51:25 > MMU_STATS COUNTER=servo_down INCR=1
18:51:25   DEBUG: Syncing gear stepper and extruder
18:51:25     TRACE: Updated sync multiplier: 1.0000
18:51:25     TRACE: Setting gear motor rotation distance: 22.724822 (ratio: 1.000000)
18:51:25     TRACE: _ensure_safe_extruder_temperature: current_temp=241.04, paused_extruder_temp=None, current_target_temp=240.0, klipper_minimum_temp=170.0, default_extruder_temp=240.0, source=auto
18:51:26 Forming tip...
18:51:26     TRACE: Running macro: EREC_CUTTER_ACTION 
18:51:28 > _MMU_STEP_MOVE MOVE=58
18:51:28   DEBUG: Moving 'gear' motor 58.0mm...
18:51:28   DEBUG: Unsyncing gear stepper and extruder
18:51:28     TRACE: Reset sync multiplier
18:51:28     TRACE: Setting gear motor rotation distance: 22.724822 (ratio: 1.000000)
18:51:29     TRACE: User defined step move. Stepper: 'gear' moved 58.0mm, encoder measured 58.3mm (delta -0.3mm). Pos: @58.0, (63.1mm)
18:51:34 > _MMU_STEP_MOVE MOVE=-1
18:51:34   DEBUG: Moving 'gear' motor -1.0mm...
18:51:34     TRACE: User defined step move. Stepper: 'gear' moved -1.0mm, encoder measured 1.0mm (delta 0.0mm). Pos: @57.0, (64.1mm)
18:51:36 > MMU_STATS COUNTER=cutter_blade INCR=1
18:51:36 > _MMU_STEP_SET_FILAMENT STATE=2
18:51:36     TRACE: (toolhead sensor detects filament)
18:51:36 [T0] < En <........ [Ex .. (*) .. Nz] 57.0mm (e:64.1mm)
18:51:36 > _MMU_STEP_UNLOAD_GATE
18:51:36   DEBUG: Slow unload of the encoder
18:51:36     TRACE: Unloading step #1 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.5mm (delta -0.5mm). Pos: @42.0, (79.6mm)
18:51:37     TRACE: Unloading step #2 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 15.5mm (delta -0.5mm). Pos: @27.0, (95.1mm)
18:51:38     TRACE: Unloading step #3 from encoder. Stepper: 'gear' moved -15.0mm, encoder measured 1.9mm (delta 13.1mm). Pos: @12.0, (97.1mm)
18:51:38     TRACE: Final parking. Stepper: 'gear' moved 0.1mm, encoder measured 0.0mm (delta 0.1mm). Pos: @12.1, (97.1mm)
18:51:38     TRACE: (toolhead sensor detects filament)
18:51:38 [T0] < En ......... [Ex .. (*) .. Nz] UNLOADED 12.1mm (e:97.1mm)
18:51:38 > _MMU_M400
18:51:38     TRACE: After tip forming, extruder moved: 0.0mm thus park_pos calculated as 16.0mm (encoder measured 92.2mm)
18:51:38     TRACE: (toolhead sensor detects filament)
18:51:38 [T0] < En <<<<<<<<< [Ex << (*) <. Nz] -16.0mm (e:20.4mm)
18:51:38     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Forming Tip'
18:51:38     TRACE: Running macro: _MMU_POST_FORM_TIP
18:51:38     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Exiting Ext' OLD_ACTION='Unloading'
18:51:38   DEBUG: Extracting filament from extruder
18:51:38     TRACE: _ensure_safe_extruder_temperature: current_temp=240.27, paused_extruder_temp=None, current_target_temp=240.0, klipper_minimum_temp=170.0, default_extruder_temp=240.0, source=auto
18:51:38     TRACE: (toolhead sensor detects filament)
18:51:38   DEBUG: Reverse homing up to 41.4mm to toolhead sensor (synced)
18:51:41     TRACE: Reverse homing to toolhead sensor. Stepper: 'gear+extruder' did not home after moving -41.4mm (of max -41.4mm), encoder measured 0.0mm (delta 41.4mm). Pos: @-57.4, (20.4mm). Pos: @-57.4, (20.4mm)
18:51:41     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Unloading' OLD_ACTION='Exiting Ext'
18:51:41     TRACE: Running macro: _MMU_ACTION_CHANGED ACTION='Idle' OLD_ACTION='Unloading'
18:51:41 MMU issue detected whilst out of a print
         Reason: Unload sequence failed: Failed to reach toolhead sensor after moving 41.4mm.
         Occured when changing tool: T0 > T3
18:51:41   DEBUG: Setting servo to up (filament released) position at angle: 140
18:51:43     TRACE: Processing idle_timeout 'ready' event

```

mochoandre, 2024-08-28:
Ok, find the problem 
`
form_tip_macro:  EREC_CUTTER_ACTION`

I am pretty shurre I've seen in some piece of documentation that you should add it 

Solved

moggieuk, 2024-09-04:
Oh!! That would cut in the wrong place.   You still need some form of tip forming with EREC -- you can use the default or create your own dummy/simple one and set:
`form_tip_macro` to that.

The EREC_CUTTER_ACTION insertion is a post_load extension documented here:
https://github.com/moggieuk/Happy-Hare/wiki/Addon-Feature-Setup#---erec-filament-cutter

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

The early cut came from using the cutter macro as the tip-forming macro; it should be set as the post-unload extension instead.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk