moggieuk/Happy-Hare #385: _MMU_CUT_TIP does not perform retraction now

SHKinsem, 2024-08-16:
_MMU_CUT_TIP does not perform retraction now.
The cutting process worked but I cannot hear the extruder working, lead to "Unload sequence failed: No encoder movement"
This happens after I updated the version.
Not 100% sure if this is a bug, however my extruder works independently if I press the extrude bottom.

SHKinsem, 2024-08-17:
After setting "encoder_move_validation" to false, everything works flawlessly.
I think maybe there is a problem with the encoder checking logic?

moggieuk, 2024-08-19:
Thanks I will take a look at this test.  Note that retract now is handled by HH “toolchange_retraction” so removed from cutting macro

SHKinsem, 2024-08-31:
> Thanks I will take a look at this test. Note that retract now is handled by HH “toolchange_retraction” so removed from cutting macro

OK, I just tried to debug this myself.
It turned out that one cannot set the "toolhead_ooze_reduction" too high in mmu.parameters.cfg

I see that the value is stored as `printer.mmu.extruder_residual_filament`, is also used for calculating the effective retract length:
```
# Step 1 - Calculate initial retract to save filament waste, repeat to allow some cooling
    {% set effective_retract_length = retract_length - printer.mmu.extruder_residual_filament - printer.mmu.toolchange_retract %}
    {% if effective_retract_length > 0 %}
```

It will potentially lead to a negative value thus the extruder does not perform retraction cuz it did not enter the ‘if condition’
But still  I dont know why the encoder still checks the movement that does not exist

SHKinsem, 2024-09-01:
> I see that the value is stored as `printer.mmu.extruder_residual_filament`, is also used for calculating the effective retract length

However, I also don't see the reason concluding that into the effective retract length calculation in cutting process. Shouldn't the `toolhead_ooze_reduction` only be used for loading process? In that sense, in the next loading process, less filament would be loaded to prevent ooze.

moggieuk, 2024-09-03:
Thanks for digging in.  It would result in a negative value because it is only applied if positive.   I did extend the range allowed by `toolhead_ooze_reduction` after learning of how much filament can be left behind especially on HF nozzles..

closed 2024-12-15 by moggieuk