moggieuk/Happy-Hare #386: Enable Bypass with multi color print

o1hitman1o, 2024-08-17:
Is there any way to add logic to enable multi color prints when using the bypass? I would say to call the pause macro when a tool change is requested when using the bypass. I am not sure if this would go here and then cause any tool change request to be redirected to the pause command to the filament could be manually changed.

        SET_GCODE_VARIABLE MACRO=_MMU_ERROR_DIALOG VARIABLE=show_abort VALUE={True} # Show abort option during startup
        {% if using_bypass and filament_loaded %}
            RESPOND MSG="MMU Bypass selected and loaded"
            {% if num_colors > 1 %}
                SET_GCODE_VARIABLE MACRO=_MMU_ERROR_DIALOG VARIABLE=custom_msg VALUE="{custom_msg}"
                MMU_PAUSE MSG="Bypass selected for multi-color print"
            {% endif %}
        {% else %}

moggieuk, 2024-09-04:
This would be possible by customizing the `Tx` macro wrappers .. and if selector is in the bypass position then run pause logic rather than the "MM_CHANGE_TOOL" it does today.

This isn't a feature I'm motivated to build though, but would consider a PR is done cleanly...

o1hitman1o, 2024-09-08:
My solution was to just add M600 to my slicer for tool changes. This works, I not 100%% sure how it would be implemented in HH though.

closed 2024-09-08 by o1hitman1o