moggieuk/Happy-Hare #389: Pause Timeout - Toolhead Cooldown Failure

dumuzi262, 2024-08-19:
This is a new issue that has started since upgrading to the latest version of Happy Hare.  When the system pauses due to some error (such as a load failure, etc), the normal behavior is to pause and after a specific length of time, to turn off the extruder until things are fixed but leaves the bed heater on.  With the new version, when the extruder is turned off, and the thermistor hits a temperature too low for printing, klipper panics and crashes the firmware.  This puts a very limited window to resolve a problem before things fail.  As I do have intermittent issues with a few of my gates, this problem is preventing me from starting a job that will run thru the night as a pause will automatically end up a failure.

Thanks for any help you can provide on this issue!

moggieuk, 2024-08-22:
Is this is new feature of klipper?  I've never heard of this before.

Note that when in a paused state, you can use "MMU_UNLOCK" to restore heat, then "RESUME" to continue printing.

I'd need more info like an `mmu.log` or at least copy of the console message to understand this.

dumuzi262, 2024-10-07:
I just wanted to create an update on this.  I wasn't able to force the error to happen.  However, it failed again last night with the same issue.  The console logs were as follows:

```
9:41 PM
Klipper state: Shutdown
5:47 PM
Extrude below minimum temp
5:47 PM
Extrude below minimum temp
See the 'min_extrude_temp' config option for details
5:42 PM
Disabled extruder heater
5:27 PM
Unsynced MMU from extruder
5:27 PM
Run Current: 0.80A Hold Current: 0.21A
5:27 PM
Restoring MMU gear stepper run current to 100% configured
5:27 PM
Synced MMU to extruder
5:27 PM
Run Current: 0.31A Hold Current: 0.21A
5:27 PM
Modifying MMU gear stepper run current to 40% for extruder syncing
5:27 PM
[T1] > En ......... [Ex .. (t) .. Nz] UNLOADED 140.0mm (e:0.0mm)
5:27 PM
Attempting to recover filament position...
5:27 PM
Unsynced MMU from extruder
5:27 PM
Run Current: 0.80A Hold Current: 0.21A
5:27 PM
Restoring MMU gear stepper run current to 100% configured
5:27 PM
Parking toolhead at (x:50.0, y:50.0, z:12.8) for pause operation
5:27 PM
idle_timeout: Timeout set to 72000.00 s
5:27 PM
After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
5:27 PM
MMU issue detected. Print will be paused
Reason: Gate 1 is empty (and EndlessSpool on load is disabled)
Load gate, remap tool to another gate or correct state with 'MMU_CHECK_GATE GATE=1' or 'MMU_GATE_MAP GATE=1 AVAILABLE=1'.
Occured when changing tool: T5 > T1
```

In the log file, this is the error when we hit the shutdown:
Stats 212157.6: gcodein=0  mcu: mcu_awake=0.027 mcu_task_avg=0.000016 mcu_task_stddev=0.000009 bytes_write=33801890 bytes_read=9832705 bytes_retransmit=0 bytes_invalid=81 send_seq=726481 receive_seq=726481 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=63999767 mmu: mcu_awake=0.007 mcu_task_avg=0.000016 mcu_task_stddev=0.000013 bytes_write=14529410 bytes_read=11174394 bytes_retransmit=0 bytes_invalid=216 send_seq=348937 receive_seq=348937 retransmit_seq=0 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000784 adj=64001042 CB1: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000015 bytes_write=175416 bytes_read=542157 bytes_retransmit=0 bytes_invalid=0 send_seq=29210 receive_seq=29210 retransmit_seq=0 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=49999996 adj=50000141 EBBCan: mcu_awake=0.004 mcu_task_avg=0.000015 mcu_task_stddev=0.000011 bytes_write=15309844 bytes_read=5841957 bytes_retransmit=4017 bytes_invalid=0 send_seq=326539 receive_seq=326539 retransmit_seq=276373 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=64000190 adj=64000667 mmu_temp: temp=28.9  ebb: temp=32.3 heater_bed: target=60 temp=60.1 pwm=0.082 E3EZ_MCU: temp=38.1 E3EZ_Host: temp=48.4 sysload=1.17 cputime=23352.192 memavail=354740 print_time=28667.303 buffer_time=0.000 print_stall=0 extruder: target=285 temp=285.2 pwm=0.536
MCU 'EBBCan' shutdown: ADC out of range

closed 2024-11-20 by github-actions[bot]