moggieuk/Happy-Hare #392: PrusaSlicer Documentation Settings for some Edge Cases

zeplintwo, 2024-08-21:
I ran into an issue with PrusaSlicer where the sliced models were not having their extrusion multipliers applied correctly if the starting filament and any other filament required different extrusion rates.

**Edge case conditions to create the error:**

- Any filament with an extrusion multiplier other than 1 in the slicer.
- A second filament with an extrusion multiplier different from the above filament.

Slicing with stock wiki settings will execute properly, and the slicer will return valid files. Prusa assumes this is what you want and does not automatically check the extrusion multiplier per tool.

The issue results in the print being over-extruded because the slicer never checked the multiplier during slicing. It seems that it never does unless you are using only a single tool. When running multiple tools, you must tell PrusaSlicer to invoke 'M221' to apply any filament profile-based extrusion multipliers.

There are two additions needed to get these values applied to your sliced G-code. Both are added in the "Custom G-Code" section of the "Printer" tab.

### First is in "Start G-Code" 
`M221 S(extrusion_multiplier[initial_extruder]}`
 I appended it at the end of your code from [here](https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup#---start-g-code) 
```
MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! FILAMENT_NAMES=!filament_names! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK
; Enter YOUR start_print macro call here
MMU_START_LOAD_INITIAL_TOOL
; Optionally add YOUR additional start logic to run just prior to start
M221 S(extrusion_multiplier[initial_extruder]}
```

Second is in "Tool change G-code"
`M221 S{extrusion_multiplier[next_extruder]}`

I appended it before the [tool change](https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup#---tool-change-g-code) as i was not sure what if any affect it would have on purging.

```
; Retrieve the extrusion multiplier for the current filament
M221 S{extrusion_multiplier[next_extruder]}
T[next_extruder]

```

These two changes allowed me to properly get the extrusion multiplier settings to transfer from the slicer to the printer. This is an edge case, as I would expect most people to run the same filaments or filaments with nearly the same flow rates. But since it was not in troubleshooting nor in the documentation in slicing settings, I wanted to bring it up.

I love the tool you have created. Thank you.

moggieuk, 2024-09-04:
Thanks for the insights.  HH does indeed update extrusion multiplies and keeps a record per-tool.   I need to think about interaction with the slicer setting....

moggieuk, 2024-09-17:
Second update:  I starting to look into what the slicer is doing.  I can't get it to inject M220 commands automatically even for single color prints.  I wonder if it is using the slicer extrusion multiplier to alter all `G1 Exx` moves?
I've never looked at this closely before wondering if you can shed some light.
@zeplintwo

zeplintwo, 2024-09-17:
That is indeed an unusual behavior. In my case, I must manually insert the M221 command into the G-code for each tool change, as it does not seem to apply the appropriate 'multipliers' otherwise. I haven't interacted much with M220, as according to the RepRap Wiki, it pertains to the speed multiplier, which doesn't seem related to my issue.

Recently, I switched to Orca Slicer, primarily because of its 'seam wrapping' feature that better hides the layer seams. However, this issue with the extrusion multiplier seems to have persisted since the switch, leading me to believe it may be a configuration error on my side. Specifically, the default G1 Exx extrusion control for the multiplier does not appear to be functioning as expected. To ensure proper compensation with Klipper, I have had to manually append M221 in my custom G-code setup.

While it’s possible I have configured something incorrectly, it’s worth noting that I have been setting up slicer profiles for over 20 years, so I might be missing something subtle. I will revisit some of the Ellis guides on extrusion multipliers within slicer settings to verify if I am overlooking any crucial details. Alternatively, if there is a more recommended guide for tuning these printers, I would appreciate any suggestions.