moggieuk/Happy-Hare #393: Set new filament temp before or directly after unloading old filament

newtop95, 2024-08-21:
I am running an ERCFv2 on a voron 2.4 with blobifier. 
The problem I have, emerges when printing different materials with different temperature requirements, since when the new filament is loaded, it is done so with the old filaments’ temperature. 

Specifically, I get two scenarios: 

## From high temp Filament to lower temp filament
 1. The old filament is unloaded with correct temp.
 2. High Temp from old filament is still applied 
 3. New filament is loaded 
 4. After loading, new, lower temp is applied. The nozzle oozes while extruder travels to the part and the temperature lowers to the newly set, lower temperature. 
 
## From low temp Filament to hinger temp filament
 1. The old filament is unloaded with correct temp.
 2. Low Temp from old filament is still applied 
 3. New filament is loaded, since filament is loaded with a to low temp, is barely extrudes
 4. After loading, new, higher temp is applied
 5. 
 
I guess the fix would be to apply new temp while unloading?

moggieuk, 2024-09-03:
Happy Hare does not at this point use any data captured from the slicer to alter temperatures.  It is a wish list item but this would be new functionality.  HH relies 100% on slicer for temperature changes.

moggieuk, 2026-10-23:
Thanks @TinkerTonkotsu for raising this. A related improvement has now been merged in #1448 and will ship in the next release. Look out for **v4.2.0**.

Happy Hare now remembers the temperature of the filament last used in each extruder. When HH runs the purge, the load is done at the higher of that temperature and the new filament's, then drops to the new filament's temperature once the purge is complete. This applies to the initial tool load at print start and to standalone loads.

**Mid-print toolchanges:** those are still controlled by the slicer, which is the case you described. For those, set the new temperature in the slicer's toolchange gcode before the `T` command, as in the workaround earlier in this thread. If you still see problems after v4.2.0, please comment here with an `mmu.log`.

moggieuk, 2026-11-03:
Shipped in [v4.2.0](https://github.com/moggieuk/Happy-Hare/releases/tag/v4.2.0) via #1434 and #1448. When Happy Hare controls the purge, the initial load and standalone loads now heat to the higher of the last filament's temperature and the new one, then drop to the new filament's temperature once the purge is done. For mid-print toolchanges the temperature is still set by the slicer, so use the toolchange G-code workaround described earlier in this thread. Closing for now — please reopen with an `mmu.log` if you still see problems on v4.2.0.

closed 2026-11-03 by moggieuk