moggieuk/Happy-Hare #396: _MMU_CUT_TIP 'Move out of range' when unloading filament at start of a print

Belstrekkie, 2024-08-23:
Since 2.7 I'm experiencing 'Move out of range' errors at the start of a print, if filament is still loaded and HH is trying to unload filament by running the _MMU_CUT_TIP macro. The error only occurs after xy homing is initiated by the macro (at the start of a print, before normal xy homing has been done) and for me always results in: Move out of range: 680.007 680.005 0.000 [0.000].

EdwinPock, 2024-08-24:
Do you use somewhere in Gcode Skew Correction?

Belstrekkie, 2024-08-24:
Yes, but I don't think that is the issue here as issues with skew correction were solved for me with the variable skew correction option in the boblifier.cfg before and those errors (in my case) always had to do with a very small negative range. The error in this case is very large (with 680 680 the toolhead basically wants to move to my neighbors bed). Also, as I said, it only occurs when filament is still loaded at the start of a print and the _MMU_CUT_TIP macro has to perform xy homing. If I eject the filament with klipper screen happy hare edition before the print (which also starts with xy homing followed by a filament cut), there are no issues and filament changes (and the cut macro) work normally during a print this way. The _MMU_CUT_TIP macro seems completely rewritten compared to the 2.6 version which didn't cause issues, so I would guess the issue has something to do with how homing is handled by this macro.

EdwinPock, 2024-08-24:
It happened to me during the tests because I didn't have "z" at home. But there may be something else with you. And these values ​​match something? Multiple variables?

EdwinPock, 2024-08-24:
Do you have the max and min parameters correctly for the X and Y axes in the toolhead settings? Because the new macro section has been added "Fillametrix Move_in_Bounds" and control of these settings (max or min axis).

Belstrekkie, 2024-08-24:
Max and min are correctly configured for x and y. It's a 2.4 350, no idea where that 680 comes from.

moggieuk, 2024-08-29:
I need to see some log information to help.  `mmu.log` but also the console output.  I put in logic to protect against out of range because klipper often leaves the toolhead out of range after homing (!?).  Thus if a restore to previous location is executed it will fail.

Belstrekkie, 2024-08-29:
Attached are the MMU log and console output. Standing next to the printer, I can confirm that there are no issues while homing. Directly after homing the movement error pops up and than the toolhead moves towards its normal pause position near the boblifier. As a work around, I removed the call for the _FILAMETRIX_MOVE_IN_BOUNDS macro, which solves the issue for now.

[mmu (2).log](https://github.com/user-attachments/files/16792978/mmu.2.log)
[console.txt](https://github.com/user-attachments/files/16792979/console.txt)

moggieuk, 2024-09-04:
I read through this again and I think this is the key:
"Also, as I said, it only occurs when filament is still loaded at the start of a print and the _MMU_CUT_TIP macro has to perform xy homing"

The way klipper macros work (and I find it so frustrating) is that printer variable are NOT updated during a macro.  Thus you have to divide macro up into parts where you call a new macro after an operation just so that printer variables are refreshed. 

I suspect that after the homing operation, something is looked up that hasn't been refreshed...  I'll look at the macro at some point but you could experiment along these lines too.

Belstrekkie, 2024-09-04:
Could it have something to do with this line from the inbounds macro:
{% set pos = printer.gcode_move.gcode_position %}  

I'm not a programmer, so it's difficult for me to interpreted these lines, but during the start of the print, when the cut macro performs homing, which is the first movement of the printhead, is there any known gcode position to be used for 'set pos'?

Belstrekkie, 2024-09-06:
I've been reading the code for the last few days, and as I said I'm not a programmer and it's difficult for me to interpreted it, but the thing I wonder is:
Running the MMU_EJECT macro before I start a print, results in the printer homing, cutting and unloading without any issues (without any mentioning of the cut tip macro being used in the logs). Then why during the mmu print start macro the cut tip macro instead of the mmu_eject macro is used to unload remaining filament? Or, why is the inbounds macro part from the cut macro not necessary when running the mmu_eject macro (I have a hard time finding the content of this mmu_eject macro in the code)?

moggieuk, 2024-09-11:
I wonder if your printer is setup correctly?   Try this:
Find this macro in `mmu_cut_tip.cfg`.  And temporarily add the M118 debugging lines:
```
[gcode_macro _FILAMETRIX_MOVE_IN_BOUNDS]
description: Helper to move the toolhead to a legal position after homing
gcode:
    {% set vars = printer['gcode_macro _MMU_CUT_TIP_VARS'] %}
    {% set travel_speed = vars['travel_speed']|float %}

    {% set pos = printer.gcode_move.gcode_position %}
    {% set axis_minimum = printer.toolhead.axis_minimum %}
    {% set axis_maximum = printer.toolhead.axis_maximum %}
    M118 axis_minimum={axis_minimum}, axis_maximum={axis_maximum}
    {% set x = [axis_minimum.x, [axis_maximum.x, pos.x]|min]|max %}
    {% set y = [axis_minimum.y, [axis_maximum.y, pos.y]|min]|max %}
    G1 X{x} Y{y} F{travel_speed * 60}
```
Let me know what is reported in the console when you run.

Belstrekkie, 2024-09-11:
Where exactly do I put the M118? I tried at the the start and the end of the move in bound macros, but I don't think the console is showing more information then before:
[console.txt](https://github.com/user-attachments/files/16967685/console.txt)

moggieuk, 2024-09-17:
The M118 line is shown in my previous post...
```
    M118 axis_minimum={axis_minimum}, axis_maximum={axis_maximum}

```

Belstrekkie, 2024-09-17:
Next try:
[console.txt](https://github.com/user-attachments/files/17023843/console.txt)

moggieuk, 2024-09-30:
I'm going to recommend we ping the author of Blobifer (@Dendrowen ) because I'm not sure what is happening...

Belstrekkie, 2024-09-30:
Have you been able to reproduce the error? With the latest updates of HH installed, I now get one additional line of information, directly before the move out of range errors:

'Warning: Klipper reported out of range gcode position (x:340.00390625, y:340.00234375)! Adjusted to (x:340.00390625, y:340.00234375) to prevent move failure'

340 by 340 isn't out of range and this exactly half of the subsequent out of move errors (680 by 680).

I now tried again with Blobifier disabled, but that didn't solve the issue :(

closed 2024-11-14 by github-actions[bot]