moggieuk/Happy-Hare #397: filament sync tension is never corrected with two sensors set up in some cases

ntchris, 2024-08-23:
filament sync tension is never corrected with two sensors set up in some cases

Steps to reproduce:
1 printer has expand and compress sensors setup, two,  not one.  ( expand sensor for too little filament, compress sensor for too many filament )
2 manually put more filament in tube to make the compressed sensor triggered.
3 manually set the mmu_hardware.cfg file  rotation_distance to a obvious smaller value , ie 20. ( so ERCF shall push more filament than it should)

Restart raspberry PI controller computer or klipper (  sudo  service klipper restart ), things are correct, (this explains why sometimes it works sometimes it doesn't. )
**however, when we <restart firmware> from the webpage (key step: klipper software did not restart)**

**Expected behaviour:**
filament sync state get updated by actual sensor reading --  compressed,  and adjust the rotation_dist

**Actual behaviour**
sync state remains neutral ( should be compressed), 
and if we start printing, since the ERCF will keep pushing more filament than it should (by the smaller rotation_distance value in config) , the already triggered compressed sensor will remain the same, the event _handle_sync_feedback() will never happen.
thus the multiplier and gear ratio remains the same.

** Propose Fix **
read the actual sensor reading and set when software restarts. (firmware reboot)
 def handle_ready(self) or  maybe  def _enable_sync_feedback(self):
`    def _enable_sync_feedback(self):

    def _enable_sync_feedback(self):
        if self.sync_feedback_operational: return
        self.sync_feedback_operational = True
        self.reactor.update_timer(self.sync_feedback_timer, self.reactor.NOW)

        eventtime = self.reactor.monotonic()
        state_expanded = self.mmu_sensors.get_status(eventtime)[self.SWITCH_SYNC_FEEDBACK_TENSION]
        state_compression = self.mmu_sensors.get_status(eventtime)[self.SWITCH_SYNC_FEEDBACK_COMPRESSION]
        self._log_debug(f"mmu sync sensor {state_expanded=}   {state_compression=}")
        #  -1 (expanded) and 1 (compressed)
        if state_expanded and state_compression:
           self._log_error("both expanded and compressed sensor are trigger at the same time!? check hardware!")
        elif state_expanded:
            self._log_debug("filament is in expanded state")
            self.sync_feedback_last_state =-1
        elif state_compression:
            self._log_debug("filament is in compressed state")
            self.sync_feedback_last_state = 1
        else:
            self._log_debug("filament is in neutral state")
            self.sync_feedback_last_state = 0
        self._update_sync_multiplier()
 
`
[filament_sync_always_neutral.log](https://github.com/user-attachments/files/16734355/filament_sync_always_neutral.log)

I have a simple fix in place ...will submit soon.

ntchris, 2024-08-24:
https://github.com/moggieuk/Happy-Hare/pull/398
fixed by this pr

moggieuk, 2024-08-29:
Thank you for the great explanation.  It doesn't sound like an missed startup assumption with both compression and tension sensors.  There aren't many users with this configuration yet ;-)

I'll get to your PR shortly, but one thing to note in your testing..
the rotation_distance in `mmu_hardware.cfg` is only used as an initial default value.  The calibrated rotation_distance for gate 0 is stored in `mmu_vars.cfg` and overrides whatever is in the hardware config.  Non gate-0 use the calibrated value * ratio also stored in `mmu_vars` if on a ERCF design (only single gear on Tradrack).

ntchris, 2024-08-29:
thanks for getting back to me.
tested 08-29 main branch
seems the starting state is now correct
tested compressed/expanded / and neutral states with firmware restart, 
ran 
manually trigger the two tension sensors , restart firmware and run
MMU_SYNC_GEAR_MOTOR
mmu_status

states are all correct.

closed 2024-08-29 by ntchris