moggieuk/Happy-Hare #404: The switches in pre-gate are turned off when printing.

EdwinPock, 2024-08-28:
What will happen as during the printout will disconnect the pre-Gate Switches in ERCT ?
Can he detect false runout detect ?

```shell
06:38:02 > _MMU_PRINT_START
06:38:02   DEBUG: Setting servo to down (filament drive) position at angle: 101
06:38:04 > MMU_STATS COUNTER=servo_down INCR=1
06:38:04   DEBUG: Syncing gear stepper and extruder
06:38:04 Modifying MMU gear stepper run current to 80% for extruder syncing
06:38:05 Happy Hare initialized ready for print (initial tool T5 loaded)
06:38:05   DEBUG: Job State: STARTED -> PRINTING (MMU State: Encoder: Enabled, Synced: True, Paused temp: None, Resume to state: ready, Position saved: None, z_height saved: -1.00mm, pause_resume: False, Idle timeout: 600.00s)
06:39:11   DEBUG: Automatic clog detection: new min_headroom (< 5.0mm desired): 4.9mm
06:39:14 > __MMU_GATE_INSERT GATE=5
06:39:14   DEBUG: Handling insertion detected by MMU pre-gate sensor #5
06:39:39 > __MMU_GATE_RUNOUT GATE=5 DO_RUNOUT=1
06:39:39   DEBUG: Handling runout detected by MMU pre-gate sensor #5
06:39:41   DEBUG: Retracting 2.0mm
06:39:41   DEBUG: Saving toolhead gcode state and position (X:167.4 Y:109.1 Z:1.2 E:3.8) for runout
06:39:41   DEBUG: Lifting toolhead 1.7mm with 5.0mm ramp (speed:15, accel:5000)
06:39:41 Issue on tool T5
06:39:41 A runout has been detected
06:39:41 EndlessSpool checking for additional gates in Group_5 for T5...
06:39:41 MMU issue detected. Print will be paused
```

moggieuk, 2024-08-29:
I'm not sure of your question.

When not printing:
- if the pre-gate sensor detects filament it will by loaded with "autoload" (assuming you have that option enabled).

When in print:
- if the pre-gate sensor detects filament it will simply mark the gate as having filament.
- if the pre-gate sensor detects a runout then if will initiate the runout handling / endless spool if configured.

It looks from your log that the pre-gate sensor has triggered (insertion), then untriggered (runout).

I suspect a faulty pre-gate sensor.  BTW you can disable pre-gate sensors in Mainsail if you want to confirm.

EdwinPock, 2024-08-29:
To be sure, I turned off the "mmu/hardware.cfg" resultat works ok, no detection - sensors for checking/replacement.
Thanks for your help.

moggieuk, 2024-08-29:
The problem is that microswitch have such a small movement range and slight printing tolerances can cause these to misfire.  This is a known issue with ERCFv2 RC1 and I know the team are working on an upgrade in the soon to be released next update...

EdwinPock, 2024-08-29:
I can't wait for the update and I will try to put on a little bigger balls by that time :-)

closed 2024-11-12 by github-actions[bot]