moggieuk/Happy-Hare #407: Double z hop on pause. additional z hop on resume.

lokertr, 2024-08-29:
I can't figure out why I am getting these additional z hops all the sudden. It leaves me printing in the air after a resume. When I pause a print toolhead moves to parked position and then hops up 2 additional times. I noticed in the console I get output saying 'Print already paused'. So I am assuming for some reason it is pausing more than once? In addition upon resume the toolhead moves into position to resume printing but then hops up 10ish mm's and starts printing in the air. Multiple pause and resumes will cause the z height to bump up by about 10mm each time.

Everything had been working fine. I don't know if it was a recent update or something I changed and forgot. I tried completely wiping Happy Hare and starting fresh. Tried wiping mainsail and it's macros as well, No luck. If I don't include the Happy Hare client macros, pause works perfectly fine, no mysterious z height changes. If I don't use HH client macros though...I lose blobifier function.

[mmu.log](https://github.com/user-attachments/files/16803963/mmu.log)
[logs-20240829-144206.zip](https://github.com/user-attachments/files/16803965/logs-20240829-144206.zip)

moggieuk, 2024-08-31:
It obviously shouldn't do this. Two questions:
1. Can you validate this is still a problem with the lastest "v2.7.0" (main branch).  I just want to be certain before looking into it.
2. Are you using my "client macros" in `mmu/optional/client_macros.cfg` for PAUSE/RESUME, etc? (and not others that may have their own z_hop handling).

Related to #2 -- Happy Hare (the klipper extension) manages the outmost retract / z_hop and this is defined in `mmu_parameters.cfg`.

My "sequence macros" and "client_macros" have additional settings such as toolchange min z height and an additional z_hop jump. It simplifies setup to set this park_z_hop to 0 at least to start with.
```
variable_park_z_hop             : 1.0           ; Additional Z_hop (mm) when parking (works in and out of print)
variable_min_toolchange_z       : 2.0           ; The absolute minimum height for toolchange moves during print
```

I will double check but since this is setup I use and so do many others I'm pretty sure it works.

BTW. I like the Mainsail "client_macros".  It is my intension to use them as a base (I will have to modify unless I can get the changes pushed upstream) in an upcoming update so features like "pause on next layer" will be available.

lokertr, 2024-09-03:
I am on 2.7.0. I was also using your start macros in concert with the mainsail ones. Disabling yours eliminated the extra hops. 

Changing all z hop values to 0 also seemed to resolve the issue. I was then able to go back in and apply the original z hop values without re-introducing the fault. I do not know what was up with it. It was immensely frustrating. Just one of those weird things where I must have done something to bork it...and re-inserting values fixed it.

moggieuk, 2024-09-03:
Believe fixed in patch to v2.7.0.   Please force an update.

closed 2024-11-12 by github-actions[bot]