moggieuk/Happy-Hare #410: How do you start with other gates than gate 0

utzy79, 2024-08-30:
Hi All.
I am running happy hare on a tradrack and have noticed the following.
If using any other gate than gate 0, I need to set the gate 0 as the same filament as the gate i am using, the slicer is using gate 0 for all the preliminary settings. (bed temp, hotend temp etc...)

I am aiming to have all gates setup with different filament so I can just select from my slicer and print.
Please let me know if I am doing something wrong.

tomlikesrocks, 2024-09-03:
Did you config your slicer?
I set my MMU to eject at the end of the print. 
That way the printhead is empty and ready for the next job.

my start G-Code:
MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! FILAMENT_NAMES=!filament_names! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK
PRINT_START EXTRUDER=[nozzle_temperature_initial_layer] BED=[bed_temperature_initial_layer_single]
MMU_START_LOAD_INITIAL_TOOL

my end G-Code:
MMU_END
PRINT_END

utzy79, 2024-09-03:
Hi, thanks for your response,
This is not the issue, The issue is that the slicer sets the bed temp and hotend temp based on whatever filament Is in the slicer for the first gate, even if I am not using that gate at all for the print.

moggieuk, 2024-09-03:
I'm not sure I understand your issue.  HH (moonraker extension) will pull all the data for all extruders and record in HH's gate_map.    You want to heat the nozzle depending on which filament you have selected...?

utzy79, 2024-09-04:
Thanks Moggie.
See below my current start Macro. What would you change here?
I am using Orcaslicer and Klipper with a Tradrack

;### hack-fix to generate preview-images in Moonraker / fluidd / printer's display: Write fake slicer info here so that preprocess_cancellation can process. ### 
; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47

M104 S180 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S{first_layer_bed_temperature[0]} ; set final bed temp
M190 S{first_layer_bed_temperature[0]} ; wait for bed temp to stabilize
G28
G90 ; use absolute coordinates
M83 ; extruder relative mode
BED_MESH_CALIBRATE AREA_START={first_layer_print_min[0]},{first_layer_print_min[1]} AREA_END={first_layer_print_max[0]},{first_layer_print_max[1]}

MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK

M104 S{first_layer_temperature[0]} ; set final nozzle temp
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize

; Enter YOUR start_print macro call here
MMU_START_LOAD_INITIAL_TOOL
; Optionally add YOUR additional start logic to run just prior to start

utzy79, 2024-09-08:
Never mind, figured it out, thank you
Needed to change the tool [0] to [initial_tool] as below

;### hack-fix to generate preview-images in Moonraker / fluidd / printer's display: Write fake slicer info here so that preprocess_cancellation can process. ### 
; generated by SuperSlicer 1.5.0 on 2023-04-21 at 13:13:47

M104 S180 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S{first_layer_bed_temperature[initial_tool]} ; set final bed temp
M190 S{first_layer_bed_temperature[initial_tool]} ; wait for bed temp to stabilize
G28
G90 ; use absolute coordinates
M83 ; extruder relative mode
BED_MESH_CALIBRATE AREA_START={first_layer_print_min[0]},{first_layer_print_min[1]} AREA_END={first_layer_print_max[0]},{first_layer_print_max[1]}

MMU_START_SETUP INITIAL_TOOL={initial_tool} REFERENCED_TOOLS=!referenced_tools! TOOL_COLORS=!colors! TOOL_TEMPS=!temperatures! TOOL_MATERIALS=!materials! PURGE_VOLUMES=!purge_volumes!
MMU_START_CHECK

M104 S{first_layer_temperature[initial_tool]} ; set final nozzle temp
M109 S{first_layer_temperature[initial_tool]} ; wait for nozzle temp to stabilize

; Enter YOUR start_print macro call here
MMU_START_LOAD_INITIAL_TOOL
; Optionally add YOUR additional start logic to run just prior to start

closed 2024-09-08 by utzy79