moggieuk/Happy-Hare #411: MMU CUT TIP

Hobbss27, 2024-08-30:
Hey guys, i am running a Tradrack with Happy Hair, i have a tool head cutter but it needs to be activated by using a y movement not an x movement. looking at the cut tip variables in mmu_macro_vars.cfg seems to only support an X move not a Y move. does anyone know how i could change this?

ningpj, 2024-08-30:
I haven't checked the code but wonder if you can simply specify a negative cut distance.  Should be safe enough to test by changing cut park position to the middle of the bed so you don't risk hitting something

Hobbss27, 2024-08-30:
Still would be an X move vs a Y move correct?

Hobbss27, 2024-08-30:
[image]

ningpj, 2024-08-30:
> Still would be an X move vs a Y move correct?

Sorry, my bad. I didnt read your question properly and thought you wanted to depress & cut in the opposite direction/side rather than Y.  Yes you are going to have to change the code to flip the cut direction around.

Hobbss27, 2024-08-30:
i have to redo the cutter cfg than basically make my changes, correct. i would need to reference my new section in other sections correct?

Hobbss27, 2024-09-03:
is there a way to do this. i understand how to change the macro because its basically just changing an X to a Y but how will i rereference everything.?

moggieuk, 2024-09-03:
There is a PR for y-axis cutting for the cut macro ... I just haven't got to looking at it yet.

moggieuk, 2024-09-04:
See this comment on PR 413: https://github.com/moggieuk/Happy-Hare/pull/413#issuecomment-2328011267