moggieuk/Happy-Hare #415: After Blobifier Nozzle wipe -> Z height goes near 0

Anollya, 2024-09-02:
last HH update (v2.7.0-153-g80e5b7b2):
After Blobifier Nozzle wipe -> z height goes near 0 and the nozzle scratches over the surface, when the nozzle arrives the wipe tower, the z-axis adjust the right height.

I replace the mmu_sequence.cfg with a older one, now its fixed, the z height move to the right height not to 0 after nozzle wiping.

https://github.com/user-attachments/assets/3686ab42-702a-474e-9e14-79857618c2d4

moggieuk, 2024-09-03:
Fixed. Force an update and this shouldn't happen.  It only occurs if you have no parking defined and use a wipe tower.

Anollya, 2024-09-03:
Hello moggieuk,

now its working fine.
Thank you for your quick response.

moggieuk, 2024-09-04:
Cool. Sorry about that.  Always a risk with a new release that restructures the code a little...

closed 2024-09-05 by moggieuk