moggieuk/Happy-Hare #417: Cold Pull macro error

troypouliot, 2024-09-03:
When attempting to run the ```MMU_COLD_PULL``` macro, I get the following error.

```Error evaluating 'gcode_macro MMU_COLD_PULL:gcode': UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 568: ordinal not in range(128)```

HappyHare version v2.7.0-163-g553537f5

moggieuk, 2024-09-03:
What version of python are you running?   This is the degree symbol and should be ok...?

troypouliot, 2024-09-03:
Python version is 2.7.16

Yes, I just updated HH this morning. Then again a few minutes ago with your latest release.

troypouliot, 2024-09-03:
Short term, I removed all the ```\u00B0C``` from the macro and it is currently running.

troypouliot, 2024-09-03:
Sorry, I take that back, now it is having trouble with the ```_MMU_STEP_MOVE```

```Internal error on command:"_MMU_STEP_MOVE"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/pi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/pi/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/pi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
  File "/home/pi/klipper/klippy/extras/mmu.py", line 4366, in cmd_MMU_STEP_MOVE
    self._move_cmd(gcmd, "User defined step move")
  File "/home/pi/klipper/klippy/extras/mmu.py", line 5614, in _move_cmd
    return self._trace_filament_move(trace_str, move, speed=speed, accel=accel, motor=motor, sync=sync, wait=wait)
  File "/home/pi/klipper/klippy/extras/mmu.py", line 5360, in _trace_filament_move
    self.mmu_toolhead.move(pos, speed)
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/home/pi/klipper/klippy/extras/mmu.py", line 5527, in _wrap_sync_mode
    self.mmu_toolhead.sync(prev_sync_mode)
  File "/home/pi/klipper/klippy/extras/mmu_toolhead.py", line 185, in sync
    self.unsync()
  File "/home/pi/klipper/klippy/extras/mmu_toolhead.py", line 257, in unsync
    self.inactive_gear_steppers.clear()
AttributeError: 'list' object has no attribute 'clear'```

I'm going to assume it is something on my end. I'll go back to a manual cold pull.

moggieuk, 2024-09-03:
You really need to update to Python 3.    Python 2 was end-of-lifed January 1st, 2020!!

moggieuk, 2024-09-03:
Python2 fix in mmu_toolhead has been pushed..

closed 2024-09-17 by moggieuk