moggieuk/Happy-Hare #418: Error running _MMU_ACTION_CHANGED: Error on 'SET_LED LED=mmu_leds INDEX=18  INDEX must have maximum of 17

Sparkss, 2024-09-04:
When upgrading to HH 2.7.0 I run into this error when trying to home or do any MMU activities.

Error running _MMU_ACTION_CHANGED: Error on 'SET_LED LED=mmu_leds INDEX=18  INDEX must have maximum of 17

I did not change my mmu led settings from the previous HH version, but I did set the animations_enable to 'false' and removed the symlink for led_effects.py from klippy/extras. 

[neopixel mmu_leds]
pin = mmu:MMU_NEOPIXEL
chain_count = 17
color_order = GRBW

[mmu_leds]
num_gates = 8
led_strip = neopixel:mmu_leds
exit_range = 9-16
entry_range = 8-1
status_index = 17
frame_rate = 8

  DEBUG: Filament should be out of extruder
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17
Error running _MMU_ACTION_CHANGED: Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.8 GREEN=0.8 BLUE=0.8 TRANSMIT=0': INDEX must have maximum of 17

MMU filament sensor mmu_pre_gate_0: insert event detected, Eventtime 5991.57
> __MMU_GATE_INSERT GATE=0
  DEBUG: Handling insertion detected by MMU pre-gate sensor #0
> MMU_PRELOAD GATE=0
Preloading filament in Gate 0
Preloading filament in Gate 0
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17
Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17
Error running _MMU_ACTION_CHANGED: Error on 'SET_LED LED=mmu_leds INDEX=18 RED=0.2 GREEN=0.2 BLUE=0.2 TRANSMIT=0': INDEX must have maximum of 17

Did I miss a required step for migrating to 'static' LED support? Thanks.
[klippy(28).log](https://github.com/user-attachments/files/16872309/klippy.28.log)

moggieuk, 2024-09-04:
Does it work with effects turn on?
Does it work if you disable the status LED (i.e. comment out in the mmu_hardware)?

I suspect the problem is with the static status LED which I don't have on my test MMU...  if you can narrow down I can probably find the issue.

Sparkss, 2024-09-05:
Updated to the latest patch/version and tested. No more errors during operation and everything looks good. I will continue to test but am closing this issue as Fixed. Thank you!

closed 2024-09-05 by Sparkss