moggieuk/Happy-Hare #422: Python error in MMU_SOAKTEST_LOAD_SEQUENCE

theodiem, 2024-09-14:
I was bored and decided to trigger the cmd `MMU_SOAKTEST_LOAD_SEQUENCE` , it did trigger immediate shutdown on klipper.
On klipper.log there was the following stacktrace:
```
Internal error on command:"MMU_SOAKTEST_LOAD_SEQUENCE"
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

Internal error on command:"MMU_SOAKTEST_LOAD_SEQUENCE"
Internal Error on WebRequest: gcode/script
Traceback (most recent call last):
  File "/home/dietpi/klipper/klippy/webhooks.py", line 256, in _process_request
    func(web_request)
  File "/home/dietpi/klipper/klippy/webhooks.py", line 436, in _handle_script
    self.gcode.run_script(web_request.get_str('script'))
  File "/home/dietpi/klipper/klippy/gcode.py", line 229, in run_script
    self._process_commands(script.split('\n'), need_ack=False)
  File "/home/dietpi/klipper/klippy/gcode.py", line 211, in _process_commands
    handler(gcmd)
  File "/home/dietpi/klipper/klippy/gcode.py", line 137, in <lambda>
    func = lambda params: origfunc(self._get_extended_params(params))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/dietpi/klipper/klippy/extras/mmu.py", line 6472, in cmd_MMU_SOAKTEST_LOAD_SEQUENCE
    tool = random.randint(0, self.mmu_num_gates - 1)
           ^^^^^^^^^^^^^^
AttributeError: 'int' object has no attribute 'randint'
```

Using Happy-Hare commit `151d85a`

theodiem, 2024-09-14:
I believe: 
https://github.com/moggieuk/Happy-Hare/blob/151d85a1d34fe8a6b4566b1729bb935049fa5005/extras/mmu.py#L6464
` random = gcmd.get_int('RANDOM', 0)`
Is a bit "improper", after all, random should be the python module random (as expected in line 6472) and not be assigned a int ;)

theodiem, 2024-09-14:
I fixed this way (not sure if it is consistent with the naming schema of Happy Hare, first time looking into its source):
```
dietpi@chisel:~/Happy-Hare$ git diff
diff --git a/extras/mmu.py b/extras/mmu.py
index 91173ea..71ec86f 100644
--- a/extras/mmu.py
+++ b/extras/mmu.py
@@ -6461,14 +6461,14 @@ class Mmu:
         if self._check_is_loaded(): return
         if self._check_is_calibrated(): return
         loops = gcmd.get_int('LOOP', 10)
-        random = gcmd.get_int('RANDOM', 0)
+        random_gates = gcmd.get_int('RANDOM', 0)
         to_nozzle = gcmd.get_int('FULL', 0)
         try:
             for l in range(loops):
                 self.log_always("Testing loop %d / %d" % (l, loops))
                 for t in range(self.mmu_num_gates):
                     tool = t
-                    if random == 1:
+                    if random_gates == 1:
                         tool = random.randint(0, self.mmu_num_gates - 1)
                     gate = self.ttg_map[tool]
                     if self.gate_status[gate] == self.GATE_EMPTY:
```

moggieuk, 2024-09-14:
Yeah, not the best choice of variable name.   Are you on Python2?  This appears to be fine of Python3...

theodiem, 2024-09-14:
I am on Python 3.12.
I believe it wouldn't matter since:
```
import random

random = 1
random.randint(0,10)
```
would break doesn't matter the version
```
$ podman run -it --rm docker.io/library/python:slim
Python 3.12.5 (main, Aug 13 2024, 01:30:38) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import random
>>> random.randint(0,10)
8
>>> random = 1
>>> random.randint(0,10)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'int' object has no attribute 'randint'
>>>
```

closed 2024-09-14 by theodiem