moggieuk/Happy-Hare #423: Broken Images in slicer settings wiki page

T-509, 2024-09-15:
The problem is not that there are broken images, it's that the description are useless without  them. For example "check the boxes as shown" but because there is no images so you don't know what the settings are that need to be changed.  

[image]

moggieuk, 2024-09-16:
The images are all there.  Try this link: [https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup](https://github.com/moggieuk/Happy-Hare/wiki/Slicer-Setup)

Github wiki gets confused if you use browser forward or back buttons. Use just the side bar navigation and try refreshing page if you see this.

T-509, 2024-09-16:
Strange, I've tried multiple browsers, mobile and desktop with the direct link, and 2 different computers on different networks, each come up with the same error,  inspect  returns with "406 (not acceptable)" . Which I think is client side but I'm not familiar enough, clearing cache doesn't seem to help either.  Maybe I'm just unlucky.

T-509, 2024-09-16:
cloning the wiki locally seemed to pull the images correctly wish I though of doing it earlier.

closed 2024-09-16 by T-509